commaai/openpilot #26694: Steering bug: Steering stops working at very low speeds in experimental mode.

dukeajuke, 2022-12-04:
### Describe the bug

2022 Arteon in experimental mode:

When at very low speeds or often stopping for a stop sign or another car, Openpilot displays a warning "Steering temporarily unavailable" and won't start working again even at higher speeds until the device is rebooted. At the same time, my car displays the message "Take control of the wheel" like it used to with my stock lane keeping. 

### Provide a route where the issue occurs

200d2fe234e1fef3|2022-12-03--18-50-25

### openpilot version

0.9.0

dukeajuke, 2022-12-04:
This is a better example route: 200d2fe234e1fef3|2022-12-03--16-12-04

sshane, 2022-12-05:
@jyoung8607 recommended sharing the route with him so he can look into this.

dukeajuke, 2022-12-07:
> @dukeajuke <https://github.com/dukeajuke> , happy to look into this if you're able to temporarily share your device with ***@***.*** again, just like you did when we added your fingerprint.

Thanks! I’ll do another drive with the experimental turned on and share it to you. It doesn’t happen at ever time its stops, but it does happen at least once with every drive.

dukeajuke, 2022-12-07:
Hi Jason - I shared my device with you. The drive is - 200d2fe234e1fef3|2022-12-07--12-15-05--0. The steering disconnect issue happens at the first light about half way through the drive.

jyoung8607, 2022-12-08:
@dukeajuke, I see where in the drive you encountered the steering fault, but not yet why based on the qlogs (abbreviated logs) that have been uploaded so far. At your convenience, go back to that same route in comma connect, go to Files, and Upload All Logs. That will send up some detailed debug info I need, so long as your device is on WiFi.

dukeajuke, 2022-12-09:
Logs should be uploaded now. You can look at my last drive too. It did the same thing but this time in Chill mode. Displays a warning "Steering temporarily unavailable" and it won't start working again until I shut the car off and reboot the device. Thanks in advance for looking at this!

jyoung8607, 2022-12-11:
Checked all three routes mentioned so far and unfortunately the rlogs haven't uploaded for any of them.

@sshane can you make sure these routes are queued for upload? I can't due to commaai/connect#203.

`200d2fe234e1fef3|2022-12-03--18-50-25`
`200d2fe234e1fef3|2022-12-03--16-12-04`
`200d2fe234e1fef3|2022-12-07--12-15-05`

dukeajuke, 2022-12-11:
Looks like Those drives have been overwritten already (should have bought more storage!). I’ll do another drive tonight and make sure the logs get updated.

dukeajuke, 2022-12-11:
The other routes were overwritten already. I just did another drive and the logs have been uploaded. The steering disconnects and errors out at the first light near the last 25% of the drive. - Route: 200d2fe234e1fef3|2022-12-11--17-00-49--0

dukeajuke, 2022-12-11:
17:05:13 is about where the steering error occurs.

jyoung8607, 2022-12-12:
It looks like your EPS doesn't want to see steering commands below 1 km/h. It also looks like your EPS is something new-to-us. There's an obvious difference in reported software part numbers, and a less obvious variance in its diagnostic response that makes me think it runs a different software stack than our usual. Fortunately, they should be easy to identify.

It looks like you're still running your temporary fingerprint update branch, so I've taken the liberty of pushing an update to that branch which prevents steering below 0.3 m/s (1.08 km/h). Please give that a try at your convenience.

Whether it works or not, please report back with your results. The change I've put in is more of a diagnostic measure than a final fix. If it works well, I'll add detection for this firmware to upstream openpilot. If not, I'll need to see logs again.

dukeajuke, 2022-12-12:
Thanks! I'll give it a try tomorrow. I do need to reinstall that branch to get the update correct?

jyoung8607, 2022-12-12:
Your device should already see the update and prompt you to install it from the UI. If it doesn't, go to Settings and then Software to manually check for updates. On the off chance that process doesn't work for you, yes you could reinstall.

dukeajuke, 2022-12-12:
Looks like it updated automatically overnight. The change you make fixed my steering problem! Thanks again for all your help.

dukeajuke, 2022-12-12:
Do you know if this change and the one you made to add my car's fingerprint will make it into the next release?

jyoung8607, 2022-12-12:
I'm glad the change worked for you. Re: the fingerprint, that was merged long ago and will be in the next release. Re: the steering fix, I'm still considering how best to fix it in mainline openpilot. It may be a couple days before I file a PR. This GitHub issue will remain open for now, so you can watch it for updates.

closed 2023-01-11 by sshane

otaviobps, 2023-09-30:
I'm running sunnypilot but I'm having the exact same issue. @jyoung8607 which file is responsible for tweaking this? I want to check it in sunnypilot to see if I can fix it.

Also, how can I check if my fingerprint is correct? I'm running in a 2023 VW Taos

dukeajuke, 2023-10-16:
> Hi @jyoung8607<https://github.com/jyoung8607>. Same here on a 2022 Q3. I finger printed the car myself. Just to reiterate the issue, once "Steering temporarily unavailable" shows up, it never gets cleared until i restart the car.

Yep. Still happens on mine. Usually occurs at very low speeds.

PR #26805 (2023-01-11): controlsd: set latActive with max minimum steer speed
Don't set CC.latActive (and by extension, the car port LKAS torque active control bits) if we're not truly steering.

Per discussion in #25761
Resolves #26694