commaai/openpilot #26696: power monitoring: add battery health metric

adeebshihadeh, 2022-12-04:
If we can come up with a good battery health metric, we can use it to both alert users when it's time to replace their battery and detect (unlikely) regressions that are degrading batteries.

IvanVnucec, 2022-12-16:
https://en.wikipedia.org/wiki/State_of_health

IvanVnucec, 2022-12-16:
I think this paper is a nice introduction to the SOH estimation algorithms: https://www.sciencedirect.com/science/article/pii/S2352152X22014591

IvanVnucec, 2022-12-16:
Do you have implemented State of Charge estimation? If yes, where can we found it? Also, the same question about battery temperature?

IvanVnucec, 2022-12-21:
@adeebshihadeh Did you saw my comments?

robbederks, 2023-01-10:
Before the holiday break I looked into this for a few hours (trying to estimate the capacity of the car battery from our voltage and current draw logs), but it looks like the data we collect isn't really enough to get a good estimate due to the other systems in the cars which we can't measure the draw from. Looked at a bunch of different (anti)routes from various cars, but didn't look to me like there's something significant we can do with the data we have at the moment

@IvanVnucec I think you're confused about which battery we're talking about. It's about the car's 12V lead-acid battery, not a battery contained in our devices. We do do _some_ SOC estimation by coulomb counting, mainly in https://github.com/commaai/openpilot/blob/dac7a24677ab680a0ff8fe209cc73cb14d6672a5/selfdrive/thermald/power_monitoring.py

closed 2023-12-12 by adeebshihadeh