commaai/openpilot #26713: 2022 Rav4 Hybrid "Headlight System Malfunction"

fawasme, 2022-12-06:
### Describe the bug

Using stock 0.9.0 on 2022 Rav4 Hybrid, the "Headlight System Malfunction" error will constantly display on the MID. Also the auto high/low beam switching will not work. 

Issue will appear as soon aw the car starts. Constantly. 

### Which car does this affect?

Toyota RAV4 Hybrid 2022

### Provide a route where the issue occurs

https://connect.comma.ai/b55f9c346adf3f5f/1670293977853/1670295026520

### openpilot version

0.9.0

### Additional info

Using stock 0.9.0 on 2022 Rav4 Hybrid, the "Headlight System Malfunction" error will constantly display on the MID. Also the auto high/low beam switching will not work. 
[image]
[image]

sshane, 2022-12-06:
You say constantly, but does this actually start as soon as you turn your car on? Or is it later into the drive?

sshane, 2022-12-06:
Some questions that might help debug this:
- Is this a brand new cable? I would recommend trying a new one.
  - Is the relay box and C3 new?
- Did this start just recently, or is a persistent problem from when you first got your device?

fawasme, 2022-12-06:
> I talked to Igor, our hardware guy, and he says all harnesses and relay
> boxes are rigorously tested, however a bad cable may slip through here and
> there. I would recommend trying a brand new USB C 3.1 Gen 2 cable and
> report back.

Thanks for the help Shane. I just received a confirmation email that you
guys will be shipping me a new cable for free. That sounds fair. I will
test it as soon as I get it and report back to you.

One last question regarding the 22 Rav4 Hybrid w/Denso: Do you think the
LONG will be available anytime in the future (I know it won't be right
away, but in the future) for this model?. This is something that really
makes a difference on me holding onto the device for the long run or not.
Is there hope for that?

adeebshihadeh, 2022-12-06:
Yes, longitudinal is likely to be supported in the future.

sshane, 2022-12-10:
Make sure your comma power is plugged in correctly, I'd try to re-seat that.

fawasme, 2022-12-12:
No luck yet. Reseated everything, new cable keeps working bad and even worse than the original one

fawasme, 2022-12-15:
[image]

Issue being reproduced on a another Rav4 2022 Hybrid

fawasme, 2022-12-15:
Will a cable like this work to perform another test, assuming I have two bad Usb-C cables?

https://www.bestbuy.com/site/sku/6419158.p?skuId=6419158 https://www.bestbuy.com/site/sku/6419158.p?skuId=6419158

fawasme, 2022-12-18:
I tested today with a new USB-C 3.2 cable. The problem still persists. After reporting on Discord, another member suggested that the issue might be either the Harness (which by PFD comparison seems fine) or the box. However, Comma support is either ignoring my request or I don't know what's going through their department. But I just can't hold on such an expensive device and have no answers at all. This is beyond unnaceptable, I will proceed to return everything until a fix has been stablished for this car model. You need to understand that I have a return window on a device that is NOT WORKING properly on an advertised vehicle which is supposed to work. 

I wished I had a different outcome and actually helped the project. I will not recommend this product to anyone after this experience.

fawasme, 2023-01-06:
Issue was hardware related and specific to the C3 device. Using the original harnesses, cables and  box the problem was solved by a replacement unit. This ticket can be closed.

sshane, 2023-01-06:
Great to hear! Let me know if there's any other issues!

closed 2023-01-06 by sshane