commaai/openpilot #26723: Standstill message causing car not to continue anymore

ridgewayer25, 2022-12-08:
### Describe the bug

RAV4 with a SDSU never had this issue of hitting the gas or resume to accelerate after stopped. Now the standstill message shows up and car does not continue when the lead car goes. Running master ci but someone in discord still said to report issue here. 

### Which car does this affect?

2018 Toyota rav4 hybrid

### Provide a route where the issue occurs

0fd770798a15a939|2022-12-07--13-30-43--0

### openpilot version

0.9.1/master-ci/c5a0289/dec 06

### Additional info

Experimental mode and disengage on gas were off.

adeebshihadeh, 2022-12-09:
@sshane is this due to your recent standstill logging?

sshane, 2022-12-09:
Yes, this broke auto start for this user. I assumed that NO_STOP_TIMER cars don't ever go into standstill since we don't request it (via the ACC_CONTROL), however that's not the case as seen here:

[image]

These cars actually do go into a full "standstill" state where the car holds it in place. The difference is that all we need to do is request positive acceleration (or enough to get moving based on hill pitch) and it leaves standstill. Would like to keep logging standstill since it technically is in a special state (albeit one that we can exit ourselves), but I'll revert for now and properly fix in our Prius.

sshane, 2022-12-09:
Should be fixed on master! https://github.com/commaai/openpilot/pull/26733

closed 2022-12-09 by sshane