commaai/openpilot #26726: ModuleNotFoundError: No module named 'Crypto'

makaveli10, 2022-12-08:
### Describe the bug

Hello, 
I am trying to run openpilot in carla simulator.
This is what I do:
1. ```./ubuntu_setup.sh```
this runs successfully.

2. ```scons```
Scons fails with:
```
scons: Reading SConscript files ...
Git commit hash for gitversion.h: dc780f85
ModuleNotFoundError: No module named 'Crypto':
  File "/home/openpilot/SConstruct", line 411:
    SConscript([
  File "/home/.local/lib/python3.8/site-packages/SCons/Script/SConscript.py", line 661:
    return method(*args, **kw)
  File "/home/.local/lib/python3.8/site-packages/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/.local/lib/python3.8/site-packages/SCons/Script/SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/openpilot/body/board/SConscript", line 129:
    certs = [get_key_header(n) for n in ["debug", "release"]]
  File "/home/openpilot/body/board/SConscript", line 129:
    certs = [get_key_header(n) for n in ["debug", "release"]]
  File "/home/openpilot/body/board/SConscript", line 100:
    from Crypto.PublicKey import RSA

```

### OS Version

Ubuntu 20.04

### openpilot version or commit

latest commit

closed 2022-12-08 by makaveli10

mimi89999, 2024-03-01:
I guess that you forgot to run `source .venv/bin/activate`. Maybe that should be added to https://github.com/commaai/openpilot/blob/master/docs/WORKFLOW.md ?

muzi2045, 2024-05-23:
> I guess that you forgot to run `source .venv/bin/activate`. Maybe that should be added to https://github.com/commaai/openpilot/blob/master/docs/WORKFLOW.md ?

that's right. it works for me.