commaai/openpilot #26739: nav: switch to MapLibre widget

adeebshihadeh, 2022-12-09:
Seems like it's being actively developed now: https://github.com/maplibre/maplibre-gl-native

birkskyum, 2023-07-10:
Hi, I'm part of the MapLibre Board, and wanted to give you a status on this.

MapLibre Native is indeed being being actively developed now, with staff maintenance to secure that PRs are reviewed, that CI is running and that docs are deployed. There are open technical steering committee meetings hosted monthly, and a consistent inflow of PRs from Meta, AWS and other companies investing in the renderer. It seems like you use a Qt widget - we have releases for both [Qt](https://github.com/maplibre/maplibre-native/releases/tag/qt-v2.0.1) and for [QML](https://github.com/maplibre/maplibre-native-qt/releases/tag/v2.1.0), supporting up to Qt 6.4.

The best points of contacts for technical questions are:
* The core renderer - @louwers
* Qt/QML SDK - @ntadej

The key objective for MapLibre Native in 2023 is to add support for Metal, because currently only OpenGL v3 is supported. This process is undertaken by WetDogWeather and is well underway.

birkskyum, 2023-08-06:
Watching the [talk](https://www.youtube.com/watch?v=ZEMPQjK1ZPw) made by @mitchellgoffpc at COMMA_CON 2023, it seems like some challenges hit here that can be resolved upstream in MapLibre are:
* Fixed scale, dynamically zooming in/out to account - Issue inherent to with mercator tiles. Investing in adaptive projections / Globe view can help resolve this.
* Speed limits - The new [data format](https://docs.overturemaps.org/) presented by Overture Map Foundation can help unlock this type of data, which is somewhat tricky in the OSM data format. Tracking issue for [Planetiler support](https://github.com/onthegomap/planetiler/issues/636)

adeebshihadeh, 2024-01-19:
Great, thanks for the update. I think someone will pick this up soon now that there's a bounty.

adeebshihadeh, 2024-01-19:
openpilot targets Linux

closed 2024-02-01 by adeebshihadeh

birkskyum, 2024-03-26:
@adeebshihadeh , there will be a MapLibre Navigation meeting on April 11th, at 18 CET (1 hour), if anyone from the OpenPilot team or community would like to join. It's a good opportunity to share your interests, to make sure the direction of the libraries align with your use cases.

It's completely open - and this is some bullet from a tentative agenda to give an idea what will happen:

- Give everyone who has a use case for nav a chance to share what they’re looking for and the challenges they see.
- Give everyone who’s somewhat actively working on something nav related a chance to talk about what they’re building, why, the unique use cases, constraints, etc. that they have, and so on.
- Identify areas where we’d like to see MapLibre Native improve to be more suitable for navigation (ex: camera and puck) for our various projects.
- Identify any other areas where we can consolidate efforts / collaborate.