commaai/openpilot #26742: ACC/LKAS/RDM Dashboard Error Honda Accord 2020

joseption, 2022-12-09:
### Describe the bug

Recently switched over from a Comma 2 which was running the stock version of OpenPIlot perfectly fine and without issue. When using Comma 3 with stock (release3) OpenPilot software on a non-hybrid Honda Accord 2020 the vehicle dashboard displays an ACC, LKAS, and RDM error. During this time, OpenPilot does not display any errors but does not appear to receive or respond to any of the cruise control commands from the vehicle. With the vehicle's cruise main turned on the ACC, LKAS, Honda sensing icon, and the info icon are all red. Clicking on the cruise +/- steering wheel buttons make an audible error sound and nothing happens on the OpenPilot display. This problem can be reproduced once OpenPilot loads and switches the harness box relay to use the Comma 3.

The harness, harness box, and all cables have already been reconnected and even replaced to ensure that the issue is not due to a faulty connection. I suspect this is software related specifically with my vehicle year since, during testing, switching to the master branch and enabling experimental mode does not yield any errors on either the Comma 3 or the vehicle and everything works as it should.

I would really like to get the release version working for me since I prefer using the vehicle's ACC for now. Any help is appreciated!

### Which car does this affect?

Honda Accord 2020

### Provide a route where the issue occurs

7c21724ca42488d4|2022-11-28--13-01-01--13

### openpilot version

0.9.0

sshane, 2023-01-17:
Hi @joseption, is this still an issue for you?

sshane, 2023-05-10:
Any updates @joseption?

sshane, 2023-05-10:
It looks like you returned your device, so I'm going to close this. In the original route you sent, panda was reporting issues sending messages to the camera (likely all the messages it should have been forwarding), which would explain why everything was fine when using openpilot longitudinal, as we were blocking the camera's messages displaying faults.

Likely hardware issue, either panda or the OBD-C cable connected to the comma three.

[image]

closed 2023-05-10 by sshane