commaai/openpilot #26757: Disconnecting DSU no longer works. Toyota Prius 2017 TSS-P

husamia, 2022-12-10:
### Describe the bug

with DSU disconnected I start C3, wait for it to come on, then start the car. I get cruise fault, CAN errors. Below are three attempts documenting the bug

1st test
27ab25db5cea2f1f|2022-12-10--17-39-27--0
2nd
27ab25db5cea2f1f|2022-12-10--19-34-21--0
3nd test
27ab25db5cea2f1f|2022-12-10--19-36-53--0

Ref: https://discord.com/channels/469524606043160576/524327905937850394/1051226947293823037

there are others in the channel that have the same problem for more context see 
https://discord.com/channels/469524606043160576/524327905937850394/1050088618082963557

### Provide a route where the issue occurs

27ab25db5cea2f1f|2022-12-10--19-36-53--0

### openpilot version

0.9.0

adeebshihadeh, 2023-04-19:
Is this still a problem?

sezniko, 2023-04-19:
Yes my 2020 Prius throws errors when unplugging the DSU.

sshane, 2023-07-24:
Duplicate of https://github.com/commaai/openpilot/issues/26624, you can try the linked branch to see if it fixes your issue

closed 2023-07-24 by sshane