commaai/openpilot #26771: 2020 Silverado Wanders straight, hugs curves

solardude2, 2022-12-12:
### Describe the bug

Wanders lane straight, hugs lines straight, hugs inside curves and crosses lines in curves. Has been doing this since being supported.  View of road and 360 camera showing lane placement can be seen here.  Unlisted video (https://www.youtube.com/watch?v=xU4Cuonmq_s)

### Which car does this affect?

2020 Silverado

### Provide a route where the issue occurs

38aa7da107d5d252|2022-12-11--11-05-23

### openpilot version

 openpilot v0.9.1

morrislee, 2022-12-12:
Same here from Sierra, the lane cutting on curves seems to be happening other cars too, there are a few reports even on small cars.

sshane, 2022-12-12:
@nuwandavek hopefully your Bolt tune should extend to all GM

Verylukyguy, 2022-12-12:
Has there been any progress on the Bolt/GM tuning?

morrislee, 2022-12-14:
I believe they are referring to https://github.com/commaai/openpilot/pull/26371

sshane, 2023-05-10:
@solardude2 does this still happen on latest master?

solardude2, 2023-05-10:
Last time I used it. Nobody with this type truck is using master. All on forks. I went back to my old tried n true and the other guys are on a fork of master trying to get it to work better. I will happily switch back for testing. Cuts curves and doesn't stay straight.

sshane, 2023-08-08:
@nuwandavek does the non-linear function we have work for this car?

morrislee, 2023-08-08:
NNFF from twilsonco is very much of an improvement for the oscillation in straights and over cutting in turns. But nothing so far fixes hugging in laneless profile. Lane planner still exist in sunnypilot has slightly better reliability in centering. I dont know if its a planner cost issue needs to be tuned or model is natually hugging when its still a taller vehicle

morrislee, 2023-08-08:
**Lane Planner**
[image]

**Laneless Planner**
[image]

Same route with different planner...

Doesnt look like a control issue, they look identical, this is with tim's NNFF, lane planner from sunny's fork to compare.

Lane positioning is very close to the Laneless from master that tim has added NNFF in, master just more oscillation and poor turn performances

nuwandavek, 2023-08-08:
@morrislee Can you post these route ids? 
@sshane I'm working on an NNFF like controls stack for this (which should greatly improve oscillation/turns), but hugging is more upstream, usually. I can look at the logs and comment better.

morrislee, 2023-08-08:
> @morrislee Can you post these route ids? @sshane I'm working on an NNFF like controls stack for this (which should greatly improve oscillation/turns), but hugging is more upstream, usually. I can look at the logs and comment better.

Laneless:
fe80a4e1cedec853|2023-08-05--17-47-13--0

Lane Planner:
fe80a4e1cedec853|2023-08-07--18-20-15--0

sshane, 2023-08-08:
Requested upload, please keep the device on wifi

nuwandavek, 2023-08-22:
[This](https://github.com/commaai/openpilot/pull/29540) should fix a lot of the lane wandering etc issues. The tune was pretty broken previously. 
@morrislee tried it and it is better than master. The neural fit is the final answer for all controls issues, but it will take a bit of time to be done more generically. As far as non-controls turn cutting, it is upstream of controls (model-planning) and is a top priority for comma. Multiple people are working on various approaches, and will hopefully have a fix soon.

closed 2023-08-22 by nuwandavek

sshane, 2023-08-23:
Fixed by https://github.com/commaai/openpilot/pull/29540