commaai/openpilot #26778: Longitudinal tuning tool

adeebshihadeh, 2022-12-13:
Longitudinal tuning is currently difficult due to the lack of repeatability and reproducibility. Let's make a tool that allows us to specify maneuvers and repeat them reliably, input at a similar place in openpilot to the joystick. With this tool, the workflow becomes:
* make a change
* test it X times on all the maneuvers specified in the tool
* check the plots with a nice PlotJuggler layout

sshane, 2022-12-13:
@ntegan1 expressed interest in this idea. Want to throw together a PR?

MoreTore, 2022-12-28:
https://github.com/commaai/openpilot/discussions/23759#discussioncomment-2638350

ntegan1, 2023-01-20:
> @ntegan1 expressed interest in this idea. Want to throw together a PR?

:)

Current status, working on cleaning up current maneuver loader/controller and then improving/tuning point of maneuver input to openpilot controls stack (see next paragraph).

Not sure whether the idea is to see how car reacts to various raw accel messages (e.g. same as joystick control) in various situations, or to replace the model in sending / feeding a plan into longcontrol. 

Right now my proof-of-concept idea replaces the long planner's job in telling longcontrol.py (and PID controller) what to do, via `a_target` and cruise set speed. (new code not uploaded but it's mostly https://github.com/ntegan1/apilot/tree/opsetspeed-v3)

Below video shows testing out manually controlling a_target (left slider) and set speed (on right) via web app.  
PJ  `ufAccelCmd` graph shows the a_target as set by phone slider when accel override is active (set speed slider < 28mph). Also set speeds below 28mph (car's min) are shown on setspeed mph graph.

https://user-images.githubusercontent.com/3820605/213632259-d6d44cf4-19f4-49e0-ad4a-a3d7de5bd35b.mp4

ntegan1, 2023-02-23:
Finished a proof of concept that takes a Maneuver and performs it while requested.
Can be performed multiple times in a row. Tested by performing the maneuver while SDSU forwarded distance button is held down.

The Maneuver is defined by an array of 3-length tuples holding velocity and acceleration values along with the time breakpoint for interpolation.

I tested out two different hardcoded maneuvers to verify aEgo and vEgo attempt to follow the maneuver/plan.

Upon maneuver request, the longitudinal plan is replaced by the maneuver interpolated into CONTROL_N / T_IDXS.

https://github.com/ntegan1/apilot/pull/16

closed 2024-10-05 by adeebshihadeh