commaai/openpilot #26813: Bolt EUV: resume from ACC standstill

jason5591, 2022-12-16:
### Describe the bug

In stop/go traffic (or at stop light), if the vehicle gets to 0mph for more than about 3 seconds, it will not resume and requires reengagement. If it is only 1-2 seconds, the vehicle will resume.

### Which car does this affect?

2023 Bolt EUV

### Provide a route where the issue occurs

776

### openpilot version

0.9.1

vanillagorillaa, 2022-12-18:
Please provide a specific route this issue occurs. You can find your uploaded routes at connect.comma.ai

sshane, 2022-12-18:
This isn't really a bug of openpilot, the car's cruise control system goes into a standstill state that we can't manipulate via spamming buttons just yet. It's a limitation of the car.

sshane, 2022-12-19:
Transferring to a discussion so we have a good place to discuss possible workarounds!