commaai/openpilot #26814: No video (blue screen) when in simulator

jeroenlammersma, 2022-12-16:
### Describe the bug

When running openpilot in the CARLA simulator, the only thing I get is a blue screen (see image below).
I checked if the car spawned in the map, but that is the case (second image below).
Furthermore, I tried an complete fresh Ubuntu 20.04 and openpilot installation, but still, same result.

[image]

[image]

### OS Version

Ubuntu 20.04

### openpilot version or commit

6dc1fa255970a5976a6acc1f1b9424258ee3fcf3

### Additional info

- CPU: AMD Ryzen 9 5900x
- GPU: NVIDIA RTX 3070

incognitojam, 2022-12-16:
This is the same issue as https://github.com/commaai/openpilot/issues/26503

closed 2022-12-16 by incognitojam

jeroenlammersma, 2022-12-16:
Ah sorry, did not see this issue. Thanks for pointing me in the right direction!

zeroyang0809, 2023-02-20:
hello i meet the problem "openpilot unavailable waiting for controls to start"
this problem is familiar with your problem
do you know how to fix it ?
thx