commaai/openpilot #26819: Build fails on M1 Mac

lccatala, 2022-12-18:
### Describe the bug

Trying to build and run OpenPilot on an M1 Macbook Pro running Ventura, I get the following output when running `scons -u -j$(nproc)`:
`scons: Reading SConscript files ...
Git commit hash for gitversion.h: dc780f85
scons: done reading SConscript files.
scons: Building targets ...
arm-none-eabi-gcc -o body/board/obj/body.elf -Wl,--section-start,.isr_vector=0x08004000 -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -T/Users/luis/openpilot/body/board/stm32fx_flash.ld -std=gnu11 -fdata-sections -ffunction-sections -Wl,--gc-sections -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DALLOW_DEBUG body/board/startup_stm32f413xx.o body/board/main-body.o body/board/hal_flash-body.o body/board/hal_pwr-body.o body/board/hal_rcc-body.o body/board/hal_i2c-body.o body/board/hal_i2c_ex-body.o body/board/hal_tim-body.o body/board/hal_tim_ex-body.o body/board/hal_adc_ex-body.o body/board/hal_cortex-body.o body/board/hal_flash_ex-body.o body/board/hal_gpio-body.o body/board/hal_rcc_ex-body.o body/board/hal-body.o body/board/hal_adc-body.o body/board/hal_dma-body.o body/board/system-body.o body/board/it-body.o body/board/bldc-body.o body/board/bldc_data-body.o body/board/bldc_con-body.o body/board/util-body.o -lgcc
arm-none-eabi-gcc -o body/board/obj/bootstub.body.elf -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -T/Users/luis/openpilot/body/board/stm32fx_flash.ld -std=gnu11 -fdata-sections -ffunction-sections -Wl,--gc-sections -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DALLOW_DEBUG body/board/startup_stm32f413xx.o body/board/rsa-body.o body/board/sha-body.o body/board/hal_pwr-body.o body/board/hal_rcc-body.o body/board/hal_cortex-body.o body/board/hal_gpio-body.o body/board/hal_rcc_ex-body.o body/board/hal-body.o body/board/system-body.o body/board/it-body.o body/board/util-body.o body/board/bootstub-body.o -lgcc
/Applications/ARM/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: body/board/bootstub-body.o:/Users/luis/openpilot/body/board/defines.h:186: multiple definition of `/Applications/ARM/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: fault_status'; body/board/util-body.o:/Users/luis/openpilot/body/board/defines.h:186: first defined here
body/board/bldc-body.o:/Users/luis/openpilot/body/board/defines.h:186: multiple definition of `fault_status'; body/board/main-body.o:/Users/luis/openpilot/body/board/defines.h:186: first defined here
/Applications/ARM/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: body/board/util-body.o:/Users/luis/openpilot/body/board/defines.h:186: multiple definition of `fault_status'; body/board/main-body.o:/Users/luis/openpilot/body/board/defines.h:186: first defined here
collect2: error: ld returned 1 exit status
collect2: error: ld returned 1 exit status
clang++ -o common/gpio.o -c -std=c++1z -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -Wno-error=unused-but-set-variable -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -I/opt/homebrew/include -I/opt/homebrew/opt/openssl/include -I. -Ithird_party/acados/include -Ithird_party/acados/include/blasfeo/include -Ithird_party/acados/include/hpipm/include -Ithird_party/catch2/include -Ithird_party/libyuv/include -Ithird_party/json11 -Ithird_party/curl/include -Ithird_party/libgralloc/include -Ithird_party/android_frameworks_native/include -Ithird_party/android_hardware_libhardware/include -Ithird_party/android_system_core/include -Ithird_party/linux/include -Ithird_party/snpe/include -Ithird_party/mapbox-gl-native-qt/include -Ithird_party/qrcode -Ithird_party -Icereal -Iopendbc/can -Ithird_party/json11 common/gpio.cc
common/gpio.cc:7:10: fatal error: 'linux/gpio.h' file not found
#include <linux/gpio.h>
         ^~~~~~~~~~~~~~
clang++ -o cereal/visionipc/visionipc_pyx.o -c -std=c++1z -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -Wno-error=unused-but-set-variable -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -Wno-#warnings -Wno-shadow -Wno-deprecated-declarations -I/opt/homebrew/include -I/opt/homebrew/opt/openssl/include -I. -Ithird_party/acados/include -Ithird_party/acados/include/blasfeo/include -Ithird_party/acados/include/hpipm/include -Ithird_party/catch2/include -Ithird_party/libyuv/include -Ithird_party/json11 -Ithird_party/curl/include -Ithird_party/libgralloc/include -Ithird_party/android_frameworks_native/include -Ithird_party/android_hardware_libhardware/include -Ithird_party/android_system_core/include -Ithird_party/linux/include -Ithird_party/snpe/include -Ithird_party/mapbox-gl-native-qt/include -Ithird_party/qrcode -Ithird_party -Icereal -Iopendbc/can -I/Users/luis/.pyenv/versions/3.8.10/include/python3.8 -I/Users/luis/Library/Caches/pypoetry/virtualenvs/openpilot-dMGGDXtI-py3.8/lib/python3.8/site-packages/numpy/core/include cereal/visionipc/visionipc_pyx.cpp
clang++ -o common/clock.o -c -std=c++1z -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -g -fPIC -O2 -Wunused -Werror -Wshadow -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -Wno-error=unused-but-set-variable -DGL_SILENCE_DEPRECATION -DSWAGLOG="\"common/swaglog.h\"" -Wno-#warnings -Wno-shadow -Wno-deprecated-declarations -I/opt/homebrew/include -I/opt/homebrew/opt/openssl/include -I. -Ithird_party/acados/include -Ithird_party/acados/include/blasfeo/include -Ithird_party/acados/include/hpipm/include -Ithird_party/catch2/include -Ithird_party/libyuv/include -Ithird_party/json11 -Ithird_party/curl/include -Ithird_party/libgralloc/include -Ithird_party/android_frameworks_native/include -Ithird_party/android_hardware_libhardware/include -Ithird_party/android_system_core/include -Ithird_party/linux/include -Ithird_party/snpe/include -Ithird_party/mapbox-gl-native-qt/include -Ithird_party/qrcode -Ithird_party -Icereal -Iopendbc/can -I/Users/luis/.pyenv/versions/3.8.10/include/python3.8 -I/Users/luis/Library/Caches/pypoetry/virtualenvs/openpilot-dMGGDXtI-py3.8/lib/python3.8/site-packages/numpy/core/include common/clock.cpp
cythonize common/kalman/simple_kalman_impl.pyx
ar rc common/libgpucommon.a common/clutil.o
scons: *** [body/board/obj/bootstub.body.elf] Error 1
scons: *** [body/board/obj/body.elf] Error 1
cythonize common/params_pyx.pyx
ranlib common/libgpucommon.a
1 error generated.
scons: *** [common/gpio.o] Error 1
Compiling /Users/luis/openpilot/common/kalman/simple_kalman_impl.pyx because it changed.
[1/1] Cythonizing /Users/luis/openpilot/common/kalman/simple_kalman_impl.pyx
Compiling /Users/luis/openpilot/common/params_pyx.pyx because it changed.
[1/1] Cythonizing /Users/luis/openpilot/common/params_pyx.pyx
scons: building terminated because of errors.`

### OS Version

Mac OS Ventura 13.0.1

### openpilot version or commit

66bdd8b12

mails738, 2022-12-20:
Did you get? Have same problem.

Countermeasure76, 2022-12-25:
I have run into the same issue on Ubuntu, and I am a bit surprised to see the fault_status struct defined as it is where it is and expected more people to have run into it since it was added ~3 months ago. I worked around the issue by removing its global scope, instead its defined in /board/main.cpp and passed into the three functions that utilize it.

lccatala, 2022-12-25:
> I have run into part of this issue on Ubuntu, and I am a bit surprised to see the fault_status struct defined as it is where it is and expected more people to have run into it since it was added ~3 months ago. I worked around the issue by removing its global scope, instead its defined in /board/main.cpp and passed into the three functions that utilize it.

What do you mean by passing it to the functions that utilize it? Can you "pass" a struct definition as a parameter?

Countermeasure76, 2022-12-25:
> What do you mean by passing it to the functions that utilize it? Can you "pass" a struct definition as a parameter?

You can pass in a struct by pointer. Not in front of my computer otherwise I'd provide a function definition change as an example.

That said, it seems like many other people can build without this issue so it's not clear to me why a few of us hit this issue. Although I would not have designed the original change to put that struct in defines.h like that from personal code practices. All the others in there are typedef'ed as true definitions which makes sense to me. This one is not and is meant to be a global, but I don't think that is the intention of that header.

jdnvn, 2022-12-29:
> You can pass in a struct by pointer. Not in front of my computer otherwise I'd provide a function definition change as an example.
>
> That said, it seems like many other people can build without this [...]

do you mind providing an example of your workaround?

adeebshihadeh, 2022-12-30:
Duplicate of #25112

closed 2022-12-30 by adeebshihadeh