commaai/openpilot #26842: Chrysler False Positive Forward Collision Warnings

ns3284, 2022-12-27:
### Describe the bug

Will be driving with Comma 3 enabled, and seemingly randomly, the Forward Collision warning will go off. It startles everyone in the car, and does not appear to have reason. Has been going on for a while on the master-ci branch.

### Which car does this affect?

2022 Chrysler Pacifica Hybrid 

### Provide a route where the issue occurs

4395faadf0a373f2|2022-12-26--22-49-31--48

### openpilot version

0.9.1 nightly

### Additional info

4395faadf0a373f2|2022-12-26--13-10-38--73
4395faadf0a373f2|2022-12-26--13-10-38--81
4395faadf0a373f2|2022-12-26--13-10-38--86
4395faadf0a373f2|2022-12-26--13-10-38--93

Here are a couple instances on a single drive. If you need more examples, I can grab more.

Verylukyguy, 2022-12-27:
I have been experiencing these False FCWs in my vehicle too, in my 2018 GMC Acadia.

My wife said that she had at least six on her way home from work yesterday.

df51b2195de66a9b|2022-12-26--16-32-25--0

adeebshihadeh, 2023-02-01:
> I have been experiencing these False FCWs in my vehicle too, in my 2018 GMC Acadia.
> 
> My wife said that she had at least six on her way home from work yesterday.
> 
> df51b2195de66a9b|2022-12-26--16-32-25--0

@Verylukyguy that route doesn't have any FCWs

Verylukyguy, 2023-02-01:
These are factory FCWs being triggered in the vehicle, not on the device; we get them almost every drive, along with seemingly random dash errors.
I am loading a current fork of Comma Master now to see if they are still occurring.

adeebshihadeh, 2023-02-02:
@HaraldSchafer I disabled radar parsing for now (#27190) since it seems the planner FCW has some [thresholds](https://github.com/commaai/openpilot/blob/cb50244a89fde385f55b2f6ad3ecd83c9b8f5c3a/selfdrive/controls/lib/longitudinal_planner.py#L129) tuned for Honda/Toyota radars. Leaving this open if you want to still take a look anyway.

HaraldSchafer, 2023-02-17:
The radar looks super noisy. Are we sure we are parsing it correctly? It frequently seems to detect slow but not quite stationary objects. The combination of not having a radar point on the true lead and having faulty non-stationary points can cause this.

adeebshihadeh, 2023-02-17:
Added a comment about why this is disabled https://github.com/commaai/openpilot/commit/dad948983a7ff26ffc073c536bd1e99b5d826cec

closed 2023-02-17 by adeebshihadeh