commaai/openpilot #26914: Fix false positive steering pressed

adeebshihadeh, 2023-01-10:


sshane, 2023-02-09:
Posted on this pull request for the results of before and after the steering pressed filter was added for Hyundai. Reduces steering pressed events with very short duration about 5x: https://github.com/commaai/openpilot/pull/26924

sshane, 2023-02-09:
```python
Steering pressed - Last 30 days - release3

Brand: nissan, dongles: 22, segments: 1618
Duration: 1 frames. Instances: 362. Instances per segment: 0.22
Duration: 2 frames. Instances: 528. Instances per segment: 0.33
Duration: 3 frames. Instances: 575. Instances per segment: 0.36
Duration: 4 frames. Instances: 613. Instances per segment: 0.38
Duration: 5 frames. Instances: 575. Instances per segment: 0.36

Brand: honda, dongles: 219, segments: 17978
Duration: 1 frames. Instances: 57065. Instances per segment: 3.17
Duration: 2 frames. Instances: 40585. Instances per segment: 2.26
Duration: 3 frames. Instances: 30899. Instances per segment: 1.72
Duration: 4 frames. Instances: 27113. Instances per segment: 1.51
Duration: 5 frames. Instances: 20706. Instances per segment: 1.15

Brand: chrysler, dongles: 63, segments: 4021
Duration: 1 frames. Instances: 43915. Instances per segment: 10.92
Duration: 2 frames. Instances: 37914. Instances per segment: 9.43
Duration: 3 frames. Instances: 24690. Instances per segment: 6.14
Duration: 4 frames. Instances: 18952. Instances per segment: 4.71
Duration: 5 frames. Instances: 9313. Instances per segment: 2.32

Brand: toyota, dongles: 709, segments: 48693
Duration: 1 frames. Instances: 6570. Instances per segment: 0.13
Duration: 2 frames. Instances: 255255. Instances per segment: 5.24
Duration: 3 frames. Instances: 9348. Instances per segment: 0.19
Duration: 4 frames. Instances: 143880. Instances per segment: 2.95
Duration: 5 frames. Instances: 4333. Instances per segment: 0.09

Brand: hyundai, dongles: 144, segments: 8419
Duration: 1 frames. Instances: 42039. Instances per segment: 4.99
Duration: 2 frames. Instances: 44616. Instances per segment: 5.3
Duration: 3 frames. Instances: 35773. Instances per segment: 4.25
Duration: 4 frames. Instances: 23674. Instances per segment: 2.81
Duration: 5 frames. Instances: 13971. Instances per segment: 1.66

Brand: subaru, dongles: 35, segments: 2327
Duration: 1 frames. Instances: 109. Instances per segment: 0.05
Duration: 2 frames. Instances: 11198. Instances per segment: 4.81
Duration: 3 frames. Instances: 189. Instances per segment: 0.08
Duration: 4 frames. Instances: 8188. Instances per segment: 3.52
Duration: 5 frames. Instances: 140. Instances per segment: 0.06

Brand: gm, dongles: 47, segments: 2536
Duration: 1 frames. Instances: 3. Instances per segment: 0.0
Duration: 2 frames. Instances: 7. Instances per segment: 0.0
Duration: 3 frames. Instances: 4. Instances per segment: 0.0
Duration: 4 frames. Instances: 2. Instances per segment: 0.0
Duration: 5 frames. Instances: 5. Instances per segment: 0.0

Brand: volkswagen, dongles: 36, segments: 2021
Duration: 1 frames. Instances: 11253. Instances per segment: 5.57
Duration: 2 frames. Instances: 7185. Instances per segment: 3.56
Duration: 3 frames. Instances: 5686. Instances per segment: 2.81
Duration: 4 frames. Instances: 4124. Instances per segment: 2.04
Duration: 5 frames. Instances: 2529. Instances per segment: 1.25

Brand: mazda, dongles: 7, segments: 260
Duration: 1 frames. Instances: 956. Instances per segment: 3.68
Duration: 2 frames. Instances: 1132. Instances per segment: 4.35
Duration: 3 frames. Instances: 897. Instances per segment: 3.45
Duration: 4 frames. Instances: 664. Instances per segment: 2.55
Duration: 5 frames. Instances: 374. Instances per segment: 1.44
```

sshane, 2023-02-09:
Here are plots for all the cars on release3, last 30 days, Nissan looks the best. x is a logarithmic scale for number of frames of the steering pressed event. https://imgur.com/a/ertaYxs

closed 2024-04-15 by adeebshihadeh