commaai/openpilot #26927: GM: get VIN

sshane, 2023-01-12:
You simply need to allow a different subset of addresses to send to in ELM mode.

varshneydevansh, 2023-08-07:
Hi Shane,

Can I pick this challenge and if possible, can you point me the initial file where to begin as I am new to this project?

leoforney, 2023-11-03:
I think this might have to do with getting VIN on GM:
https://docs.google.com/spreadsheets/d/1qEwOXSr3bWoc2VUhpuIam236OOZxPc2hxpLUsV0xkn0/edit#gid=2
Row 162 and 163

varshneydevansh, 2023-11-05:
@leoforney Thank you so much. I will start working on this in December. Currently working on my writing thing. =)

garrettpall, 2024-01-22:
@sshane What's wrong with using VIN1 and VIN2 signal?

sshane, 2024-01-23:
Right now openpilot generates carParams once on startup, this requires a bit of a change to the architecture of openpilot. Doable, and we might want to do this since other cars also publish their VIN like this, but with the Bolt at least, we can just query the camera as the easiest way to get the VIN, right now.

sshane, 2024-01-30:
Done in https://github.com/commaai/openpilot/pull/31224

closed 2024-01-30 by sshane