commaai/openpilot #26935: micd sometimes crashes on startup

sshane, 2023-01-13:


MoreTore, 2023-01-24:
Are you guys able to reproduce this error reliably? I only encounter this issue sometimes after rebooting.

adeebshihadeh, 2023-01-24:
happens when you start too quickly. easy to repro with:
```
diff --git a/launch_openpilot.sh b/launch_openpilot.sh
index 1525e1715..c6580c1a4 100755
--- a/launch_openpilot.sh
+++ b/launch_openpilot.sh
@@ -1,5 +1,10 @@
 #!/usr/bin/bash

+echo "starting micd"
+system/micd.py
+echo "all done $?"
+sleep infinity
+
 export PASSIVE="0"
 exec ./launch_chffrplus.sh
```

MoreTore, 2023-01-24:
ok so this is probably due to the driver not being ready when sounddevice is started. https://github.com/spatialaudio/python-sounddevice/issues/125.

My quick fix is just switching to portaudio server using 'default' instead of 'sysdefault' or waiting for 'sysdefault' to become available by checking `sd.query_devices(device=None, kind=None)`.

MoreTore, 2023-01-30:
Ok so for now I just added time.sleep(2) until you fix the issue.

incognitojam, 2023-01-31:
@MoreTore feel free to try out the `micd-input-device` branch

PR #27160 (2023-01-31): fix micd input device
closes #26935

- removed "sysdefault" and got a different error
- seems related: https://github.com/spatialaudio/python-sounddevice/issues/147
- removing import from top level seems to fix
closed 2023-01-31 by incognitojam