commaai/openpilot #26956: Car docs diff bot: use generated file

sshane, 2023-01-14:
Will simplify things a lot. We might lose detection when we rename a platform, but that's fine

sshane, 2023-05-23:
We can't use the generated CARS.md for everything (like the detail sentence), so looks like keeping the pickling is the only option for now (unless we make a template for site things like detail sentences?). Or maybe just add the sentence as a tooltip if possible on each car.

mowoe, 2024-01-16:
Hi, is there a reason why this was implemented in a python script in the first place? Are those scripts needed somewhere else or could the whole thing be done in pure GitHub actions?

sshane, 2024-01-16:
No, they are not used anywhere else. I think a big reason is readability and having static analysis. If it's as readable inside the Action, then sure.

mowoe, 2024-01-17:
Thanks for the clarification. One last question: In the first comment you mention that it is not possible to rely solely on the generated markdown for a diff. Why is that? 
The detail sentence is just generated from all info that is already present in the table or am i missing something here?
Generated detail sentence for the first car is
`openpilot upgrades your Acura ILX with automated lane centering and adaptive cruise control while driving above 25 mph. Traffic light and stop sign handling is also available in [Experimental mode](https://blog.comma.ai/090release/#experimental-mode).`
All of the information in there is also stored in the actual table:
|Make|Model|Supported Package|ACC|No ACC accel below|No ALC below|Steering Torque|Resume from stop|<a href="##"><img width=2000></a>Hardware Needed<br>&nbsp;|Video|Details|
|---|---|---|:---:|:---:|:---:|:---:|:---:|:---:|:---:|:---:|
|Acura|ILX 2016-19|AcuraWatch Plus|openpilot|25 mph|25 mph|[image]|[image]|<details><summary>Parts</summary><sub>- 1 Honda Nidec connector<br>- 1 RJ45 cable (7 ft)<br>- 1 comma 3X<br>- 1 comma power v2<br>- 1 harness box<br>- 1 mount<br>- 1 right angle OBD-C cable (1.5 ft)<br><a href="https://comma.ai/shop/comma-3x.html?make=Acura&model=ILX 2016-19">Buy Here</a></sub></details>||<details><summary>Details</summary><sub>openpilot upgrades your <strong>Acura ILX</strong> with automated lane centering and adaptive cruise control <strong>while driving above 25 mph</strong>. Traffic light and stop sign handling is also available in <a href='https://blog.comma.ai/090release/#experimental-mode' target='_blank' class='link-light-new-regular-text'>Experimental mode</a>.</sub></details>|

(You see that i already tried to add the decription sentence as an expandable tooltip to the table, but only later realised that all of the info is already there)

sshane, 2024-01-18:
Sure, but that requires parsing the columns and re-constructing it in two places (the original car docs and this differ). We don't want to have to update two places when changing the docs

ColsonXu, 2024-01-31:
@mowoe I want to give this a stab. Are you currently working on this?

mowoe, 2024-01-31:
Hi @ColsonXu i already have put some work into it but i am not sure i correctly understand what the callenge actually is here. I ended up creating a JS github action which just posts some excerpts of the markdown table as a comment and i also modified the parts that generate the CARS.md file to include the detail sentence, but as you see in the previous comment, i think that isnt really necessary, so thats why i was a bit confused.

So tl;dr, im not currently working on this, feel free to pick this up :)

ColsonXu, 2024-02-01:
@mowoe Got it. Thanks!

collinalexbell, 2024-05-15:
I'm picking this up

closed 2024-06-10 by sshane