commaai/openpilot #26959: Vehicle Parameter Identification Failed during poor traction event that didn't trigger ESP

nabeelr, 2023-01-14:
### Describe the bug

While driving in low traction conditions, during sharp corners where the vehicle understeers, openpilot will occasionally display the error Vehicle Parameter Identification Failed.

### Provide a route where the issue occurs

dd154b34dc4d8c52|2023-01-13--18-13-44--44

### openpilot version

ebe43deda5d52cb4751da9d7319cf91a295d515f

### Additional info

Also occured at dd154b34dc4d8c52|2023-01-13--18-13-44--41.

PhilWallace, 2023-02-07:
Not sure what triggers this bug. I met the same bug in PC simulation, and I solved it by deleting the identification failed alert in controlsd.py.

Emalton, 2023-05-02:
Have the same thing on my Subaru.

adeebshihadeh, 2023-08-07:
@nuwandavek can you look into this?

nuwandavek, 2023-08-07:
Will check it out

nabeelr, 2023-08-07:
Let me know if you guys need any more info, or if I can help in any way!