commaai/openpilot #27024: 2023 Kia Sportage X-line

samadv17, 2023-01-21:
### Describe the bug

It says that Dashcam mode is active and says unrecognized car when I try to use it.
It is a 2023 Kia Sportage X-line (and supports SCC).

### Which car does this affect?

2023 Kia Sportage X-line

### Provide a route where the issue occurs

56966d6ac2d0aa08|2023-01-20--21-16-46

### openpilot version

master

sshane, 2023-01-24:
Hi. I checked the 2023 Sportage and it doesn't look like it's available on the X-Line trim: https://www.kiamedia.com/us/en/models/sportage/2023/features

Are you sure your car has SCC? It's also reporting that there's no dedicated radar on your car, which is why it failed to fingerprint.

samadv17, 2023-01-24:
They never specified the trim on the comma website, and when I was discussing getting the comma with my friend who has one, he said that the features I have should work. I guess after seeing the features, I never realized my trim only offered the features, not full SCC. So, I'm guessing my only option is to return the comma, as it won't be compatible?

closed 2023-01-24 by sshane