commaai/openpilot #27027: Communication Issue

tglevrie, 2023-01-21:
### Describe the bug

Lately, during drives, I've received an orange screen saying that there's a communication issue. Disengaging and reengaging fixes the issue for the most part, and rebooting will reduce the frequency even more so. I just wanted to make a report to see if an issue could be identified. 

### Provide a route where the issue occurs

1c05783affe6641f|2023-01-20--19-06-44

### openpilot version

0.9.0

### Additional info

Makes no difference whether in experimental mode or not, if that helps. Issue happens around 19:10:15. Thanks a bunch for the help (:

sshane, 2023-01-24:
Seems like you're dropping CAN packets every once in a while. Have you tried a new cable?

tglevrie, 2023-01-24:
I have not. I will try this and hopefully it resolves the issue. Thank you very much!

sshane, 2023-01-26:
Let me know when you try it, and which cable you try

adeebshihadeh, 2023-01-26:
This is a system lag, not a cable issue. #27084 is another instance of this.

goodchoom, 2023-01-28:
After several days of not getting the alerts and updating to 0.9.1 master-ci e14fd545816b9546f2362717213ae9000bd69d27 they started going off again. This time they ocurred more frequently and earlier on than expected.

3d681b6c168ca058|2023-01-28--16-36-55--0

AreYouRlyScared, 2023-02-03:
I ran into this issue today when I went to lunch at work, as requested by @sshane, I'm posting the segment here. It should be noted a simple reboot fixed it. 

22 EUV, Running the latest `master` commit.
`3140e1c6195c880e|2023-02-02--20-30-40--2`

PR #27275 (2023-02-10): controlsd: revert CAN timeout threshold
#24366 changed the CAN socket timeout in order to prevent controlsd from running at 10Hz with no CAN packets coming in, which also lowered the threshold for a CAN socket timeout alert. This caused a slight uptick in the alert. 

closes #27027
closed 2023-02-10 by adeebshihadeh