commaai/openpilot #27177: 2022 Hyundai Santa Fe error messages

YUMISCOMMA, 2023-02-01:
### Describe the bug

While driving the car I can sometimes get vehicle error messages saying

1. Check rear cross traffic safety system
2. Check forward safety system
3. Check blind spot safety system

Auto pilot would beep, disengage & show in red
"Cruise Fault take control immediately!"

In addition, 1 out of 3 times when starting the car I would get the same vehicle error messages
And autopilot would say
" cruise Fault: restart the car to engage)

On a side note auto hold feature gets disabled as well.
[image]
[image]
[image]

[image]

### Which car does this affect?

2022 Hyundai Santa Fe limited 

### Provide a route where the issue occurs

1493c38de09a381a|2023-02-01--07-36-19--85

### openpilot version

0.9.0

### Additional info

Since I am an automotive electrical engineer I did first try to troubleshoot on my own.
First thing; I turned the engine off then restart it
And all the messages were still there until I started driving a little bit.
I do the same whenever I get the error messages and it's usually goes away after a little bit of driving.

I also hooked up my scanner to check the codes and what looks to me as a: can communication failure 

I tried unplugging the T- harness
And putting everything back to factory spec but that didn't take the error messages away until I drove a little bit.

If there is anything that I can do to help comma team figure this out I'll be more than happy to assist.

[image]

[image]

[image]

[image]

sshane, 2023-02-04:
I see you're not using openpilot longitudinal, so all those DTCs about SCC messages being invalid is very strange. I didn't see any issues with the counters for any SCC message when it faulted. Have you ever gotten these errors before openpilot? What if you put the device in dashcam mode?

YUMISCOMMA, 2023-02-05:
Thanks for getting back to me. I did not see any of these messages before I had installed my comma 3. I am going to try putting it into dash cam mode and seeing if these messages ever appear again. Also I'm pretty new to Comma and open pilot so I'm not exactly sure what longitudinal means. 

Actually I'm starting to think that something is wrong with my car. Because on the website it says that the Hyundai Santa Fe 2022 is compatible with experimental mode. But what happens when I try to turn on experimental mode it says:
 "Experimental mode is currently unavailable on this car, since the car's stock ACC is used for longitudinal control."

I would love to use experimental mode but I'm not sure if something is wrong with my vehicle or it's just not supported for some reason.
[image]
[image]

adeebshihadeh, 2023-02-05:
> Actually I'm starting to think that something is wrong with my car. Because on the website it says that the Hyundai Santa Fe 2022 is compatible with experimental mode. But what happens when I try to [...]

Your car also needs "Experimental openpilot longitudinal control" to run Experimental mode, which is only available in non-release branches. This is made more clear in the toggle text in the upcoming release. Check out https://discord.comma.ai for more info.

YUMISCOMMA, 2023-02-05:
Oh, got it.
So what could Thea's error messages mean?
I have an automotive ship in NY. We do emergency vehicle upfitting. So I familiar with different types of error messages. To me it looks like sometimes there's a glitch & that causes a communication failure in the can network.

adeebshihadeh, 2023-02-05:
You're right in that it's likely a CAN communication error. Unfortunately, the OBD-C cable can fail very subtly. Did you bend the cable while installing the device? You can try swapping out that cable or bending it a bit while the car is on (and parked) to see if the errors show up.

YUMISCOMMA, 2023-02-05:
Ok great, I'll crimp up an rj45 and run a brand new cable down to the obd2 port. Hopefully it's it'll solve this issue.

Btw I some how 2 questions on my comma. I'd love to know if I can shoot you an email with a couple of questions and you'll be able to answer those for me 🙏🏻🙏🏻

adeebshihadeh, 2023-02-05:
> Ok great, I'll crimp up an rj45 and run a brand new cable down to the obd2 port. Hopefully it's it'll solve this issue.

It's not the one running down to the comma power; OBD-C is just what we call the USB-C cable. 

> Btw I some how 2 questions on my comma. I'd love to know if I can shoot you an email with a couple of questions and you'll be able to answer those for me 🙏🏻🙏🏻

The [community discord](https://discord.comma.ai) is probably a better place for this.

YUMISCOMMA, 2023-02-05:
Okay thanks. I'll try to ask around.
Do you have any recommendations on a usb c replace cable?

sshane, 2023-03-02:
I checked your routes out in more detail and it appears that your right rear wheel speed sensor is defective. It's randomly jumping around, then going to the max signal value (which is when you see the faults). I would get this checked out at a dealer, since it's affecting ABS, traction control.

[image]
[image]
[image]

YUMISCOMMA, 2023-03-02:
Wow. Pretty impressive that you were able to crack down on this. It's interesting because a speed sensor shouldn't go bad on a brand new 2022 vehicle with 10k miles on the clock.
I'll definitely take it to a dealership to have a look.
Lately I've been having this a lot. Almost every morning when I come in the vehicle after remote starting I will see all the air messages on the dash. All I need to do is to turn the car off then back on the error messages will still be there until I drive past 8mph. But besides of that speed sensor do you think it could have to do with my battery? Because my battery is pretty low lately and I'm getting messages almost every morning "battery discharging due to external electrical devices". So bottom line, are you certain that the speed sensor is what's causing the issue or it could be the battery as well or both?

sshane, 2023-03-02:
I've never seen anything like this before, so it's hard to say what the cause is. Let me know when you go get it checked out!

sshane, 2023-03-22:
@YUMISCOMMA did you get your vehicle checked out?

YUMISCOMMA, 2023-03-22:
Thank you so much for reaching out.
At this time I'm already very frustrated with my car. I started hating her:)
I didn't go to the dealer because I know that they're going to blame a lot
of other things including my c3 & some aftermarket stuff that I've added
such as "hardwired dashcam & radar detector onboard air and air horns" even
though all my wiring and relays and crimps are done very professionally
because I do this for a living. I have a shop in New York called
"Autobuffny". You can check it up on Instagram we do lights and sirens for
emergency responders. So I'm pretty familiar with vehicle wiring and
schematics. And I definitely know the dealerships,  their hair to blame all
my aftermarket equipment. Now back to the car. I still get the error
message almost every morning when I remove start my car. But it's not
consistent. Sometimes yeah and sometimes no. What else I noticed is that
most of the time it happens when the vehicle is stopped. Such as upon
startup or a traffic on the vehicle comes to a full stop. What I usually
do, I wait until I hit a traffic light & I just restart the car. Once I hit
8 mph all the error codes go away. Most of the time it goes away till the
end of the trip but sometimes it does come back randomly. I really think
that my battery is what's causing it. Because I have a lot of accessories
Hooked up to it & it might be under 12v upon start up or low rpm's so thats
causing some computers to malfunction & trigger the errors.

Now to my main concern. I've had this twice and I almost killed myself.
"When the error message pops up I sometimes hear/feel vibration in my brake
pedal. It feels like the ABS pump is glitching. And whenever that happens
my brakes don't work! I've literally almost crashed into someone. My brake
pedal becomes hard as a rock and I cannot brake! Something is pretty damn
crazy with my car. I'm very concerned.

What do you think I should do?
The dealership's next appointment is in April.
I actually called them and told them that it's an emergency my brakes
aren't working. They told me that I don't have loaners at the moment and I
can either drop off my car at the dealership lot or wait for the next
appointment.
So I'm pretty much screwed.

Do you think you'll be able to gather some more info if I send you more
routes and logs?

Thank you so much for
 trying to help 🙏🏻🥰

Best regards
Ben

sshane, 2023-03-24:
Yeah I would definitely hold off on driving the vehicle until you can get into the dealer to check it out. You can remove the comma three if you're concerned about the dealer saying anything, it's not a permanent install as you know.

From what we know about your faulty wheel speed sensors (or some other wiring), who knows how the modules in the car might respond to that.

sshane, 2023-05-10:
Hi @YUMISCOMMA. I'm closing since this is a physical problem with your car, not openpilot, as noted above. Let me know if you have any other problems specifically with openpilot!

closed 2023-05-10 by sshane

YUMISCOMMA, 2023-05-10:
I appreciate it.
Yes you can actually close this case.
As a matter of fact I haven't received those errors for a couple of months
now. I'm not sure what caused it to stop coming. But so far so good.

Best regards.