commaai/openpilot #27206: Hyundai Tucson 2023 poor tuning

adeebshihadeh, 2023-02-03:
Waiting on more stock routes from user. 

```
8426de28ebdc65a2|2023-01-27--20-07-53
```

- [ ] Fix tuning/steer scaling
- [ ] Look into other HKG models that are potentially like this

adeebshihadeh, 2023-02-14:
Looks like it's just one FW version that needs different tuning. Temporarily removed it in #27340 until we better understand the difference.

sshane, 2023-03-24:
@adeebshihadeh looks like this camera FW version was added again under `SANTA_CRUZ_1ST_GEN`. I wonder if it also has bad tuning.

sshane, 2023-07-08:
@nuwandavek by the way, it looks like we have 4 active Tucson users in the last 90 days with these different parts: `{'1966303b764e7142', '8426de28ebdc65a2', 'bf1a6b0ba293c37b', 'a0191364609ee493'}`

sunnyhaibin, 2023-11-23:
I've been running `LAT_ACCEL = 2.9 - 3.1` and `FRICTION = 0.095` for the past few days with the 2022 Hyundai Tucson Hybrid, and it is significantly better than the current values.
- https://github.com/commaai/openpilot/pull/30513

sshane, 2024-03-08:
I would like to combine the Santa Cruz and Tucson. According to the [wikipedia page](https://en.wikipedia.org/wiki/Hyundai_Santa_Cruz), it is "closely related" to the Tucson as it shares the platform code with an extension of OB meaning Open Bed. The mass and wheel base are a bit different though.

@nuwandavek can you look at these two vehicles in the fleet and determine if using a median value is safe, letting torqued learn the difference?