commaai/openpilot #27232: Openpilot perform poorly in Carla Simulation

PhilWallace, 2023-02-06:
### Describe the bug

Hello guys. I'm running the latest Openpilot on Carla simulation, and found out it perform quite poorly... Specifically, It cannot even stay in the lane with very low speed (25 mph or even lower), and it is very likely to crash in some turnings:

[image]

Considering the good performance of Openpilot on real car, I wonder what is the reason leading to such poor performance in simulation. Maybe the parameter of the MPC control not well-tuned? Or the vehicle dynamic model of Openpilot does not fit the Carla's car model? I also felt that the detected lanes are quite laggy, which may not provide real-time information for the planning can controlling module.

 Any advice on improving the performance in Simulation? Many thanks.
 

### OS Version

Ubuntu 20.04

ferryman0723, 2023-02-08:
I met the same situation and can‘t solve it！

adeebshihadeh, 2023-12-12:
We've moved our simulator to MetaDrive and it performs much better now. Let us know if you have any issues with that.

closed 2023-12-12 by adeebshihadeh