commaai/openpilot #27243: Laikad Crash

bjsmyth, 2023-02-07:
### Describe the bug

Was driving normally when OP disengaged and reported `laikad` had crashed. Running the master branch with the latest commits at the time of the issue.

### Provide a route where the issue occurs

f8eaaccd2a90aef8|2023-02-06--17-53-50

### openpilot version

0374d48

gast04, 2023-02-08:
Lets keep this open,
the fix implemented is only temporary, as it doesnt tackle the problem, the issue is that two times the data for string number 4 is received, but the second data is totally different from the first one. The hamming code of both datas are valid (pretty sure also ublox verifies this and if the hamming code is invalid it will not forward it).
The faulty data overwrites the correct one in the buffer and we end up with an invalid ephemeris package...

```
good data:
24 24 a5 a4 00 00 13 11 ba c9 20 00 04 b3 00 01
0x2424a5a400001311bac9200004b30001

22 1203 1 4 36
nt: 1134
tau: -75053
dtau: 4
age: 0
p4 1
ft: 3
m: 1

faulty data:
25 10 0d a7 2d 40 ab f8 45 05 88 00 04 b3 00 01
0x25100da72d40abf84505880004b30001

22 1203 1 4 177
nt: 1553
tau: -557165
dtau: 7
age: 5
p4 0
ft: 11
m: 0
```

I am not sure how to properly fix this for now, and checking for invalid dayInYear is really hacky...

adeebshihadeh, 2023-06-12:
@haraschax still relevant?

haraschax, 2023-06-13:
This is caught with a generic try/catch, so I am closing.

closed 2023-06-13 by haraschax