commaai/openpilot #27254: Support for 2018 Lexus es 350

ShamGod100, 2023-02-08:
Can someone pull a request for 2018 Lexus es 350 so I can use my comma 3. Please and thank you!

sshane, 2023-02-10:
Can you post your device's dongle ID?

ShamGod100, 2023-02-11:
My dongle ID is

7a31f030957b9c85

sshane, 2023-02-11:
Looks like we don't support your platform at all! I'll get a branch ready for you to test soon.

sshane, 2023-02-17:
@ErichMoraga want to do this car port? it should be pretty simple

ErichMoraga, 2023-02-17:
@ShamGod100 Did you ever ask for help in the Discord `#toyota-lexus` channel?  Indeed it's curious we never got an ICE port for the pre-TSS2 ES Lexus vehicles.  If you don't know how to do a port, you can add me (ejmoraga@gmail.com) to your DongleID 7a31f030957b9c85 for access, etc.  I'm not sure Shane ever setup a branch for you, and is probably pretty busy too (hence him reaching out to me).

```
  ...
  LEXUS_ES = "LEXUS ES 2016" <--------- missing ICE port
  LEXUS_ESH = "LEXUS ES HYBRID 2018"
  LEXUS_ES_TSS2 = "LEXUS ES 2019"
  LEXUS_ESH_TSS2 = "LEXUS ES HYBRID 2019"
...
  LEXUS_NX = "LEXUS NX 2018"
  LEXUS_NXH = "LEXUS NX HYBRID 2018"
  LEXUS_NX_TSS2 = "LEXUS NX 2020"
  LEXUS_NXH_TSS2 = "LEXUS NX HYBRID 2020"
  ...
```

FWIW - I think support for the ES could've technically been extended back to 2016 per the **Lexus Safety System+** _introduction_ details in...
https://www.lexus.com/content/dam/lexus/documents/brochures/models/2016/My16-Lexus-ES-and-Lexus-ES-Hybrid-Brochure.pdf

ShamGod100, 2023-02-21:
No help on discord

ErichMoraga, 2023-02-21:
Where is your Discord post?  You still haven't added me to your DongleID either.

ShamGod100, 2023-02-21:
I don’t know how to that.

I’m using my comma 3 on my 2019 honda pilot
So maybe can you walk me through how to do all that. 

My discord 
Shams#4982

sshane, 2023-03-24:
@ShamGod100 are you still around? Made a PR to test: https://github.com/commaai/openpilot/pull/27683

ShamGod100, 2023-03-26:
Yes I’m here Can you send me a url ?Sent from my iPhone

ShamGod100, 2023-03-26:
I tested it and it worked on 2018 lexus es 350

No ACC below 28 mph.

sshane, 2023-03-27:
@ShamGod100 Nice, now we just need a test route. Since the route will be public, we recommend to find a public parking lot to start at (turn car off and back on), then drive for 5 minutes and park at another public parking lot. Then let me know once you've done this and we can merge your car!

ShamGod100, 2023-03-29:
Has it been merged?

sshane, 2023-03-31:
@ShamGod100 Have you got the test route yet?

ShamGod100, 2023-03-31:
Yeah how do I send the test route to u

ErichMoraga, 2023-03-31:
> Yeah how do I send the test route to u

It's just the DongleID|route you can get from...
https://useradmin.comma.ai/
or
https://connect.comma.ai/  [then click the "More info" button]

@sshane This was his latest route...
`7a31f030957b9c85|2023-03-29--14-51-26`

ShamGod100, 2023-04-02:
7a31f030957b9c85|2023-04-01--14-03-45--7

Use this route

ShamGod100, 2023-04-02:
7a31f030957b9c85|2023-04-01--14-12-51--17

Use this route

sshane, 2023-04-13:
@ShamGod100 your car is now supported in master!

closed 2023-04-13 by sshane

ShamGod100, 2023-04-15:
Is their any good forks for this vehicle

ErichMoraga, 2023-04-15:
> Is their any good forks for this vehicle

You must be kidding, but if not, use Discord.

ShamGod100, 2023-04-18:
I unplugged the DSU and its giving cruise fault restart car and after restarting still getting cruise fault message

ShamGod100, 2023-04-19:
Is this it?

https://www.etsy.com/listing/915424718/sdsu-smartdsu-for-toyota-openpilot

These are the supported models

Supported Vehicles:
Most of Toyota Vehicles
Lexus:
RX350
RX450
ES300H

ErichMoraga, 2023-04-19:
> Is this it?

That one is in Canada.  If you're closer to the Chinese source (AliExpress, etc.), get it from there.  Also, use Discord for questions like that.  It's strange for you to continue using GitHub like this.  I will not be responding to additional questions via GitHub.