commaai/openpilot #27255: 0.9.0 modeld not working and openpilot not accelarate

didpurwanto, 2023-02-08:
### Describe the bug

I have updated the openpilot to 0.9.0, but have several issues.

1. ``modeld`` not working
snapshot
```
create CUDAExecutionProvider. Please reference https://onnxruntime.ai/docs/reference/execution-providers/CUDA-ExecutionProvider.html#requirements to ensure all dependencies are met.
clocksd logcatd proclogd logmessaged modeld ui soundd locationd calibrationd torqued controlsd deleter laikad navd paramsd plannerd radard thermald statsd                                                                                                        
Onnx using  ['CPUExecutionProvider']
HERE ready to run onnx model ['input_imgs', 'big_input_imgs', 'desire', 'traffic_convention', 'features_buffer'] [[1, 12, 128, 256], [1, 12, 128, 256], [1, 100, 8], [1, 2], [1, 99, 128]]
---- HOHO ready to run onnx model ['input_imgs', 'big_input_imgs', 'desire', 'traffic_convention', 'features_buffer'] [[1, 12, 128, 256], [1, 12, 128, 256], [1, 100, 8], [1, 2], [1, 99, 128]]
-- GT  0  / SZ  393216  / ST  b''  /
Traceback (most recent call last):
  File "/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 75, in <module>
    run_loop(ort_session, tf8_input=("--use_tf8" in sys.argv))
  File "/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 48, in run_loop
    inputs.append(read(ts, (k=='input_img' and tf8_input)).reshape(shp).astype(itp))
  File "/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 21, in read
    assert(len(st) > 0)
AssertionError
```
Note that I have reinstalled libzmq package

2. openpilot not accelerate
```
frame:  engaged: False ; throttle:  0.0 ; steer(c/deg):  -0.0 0.0 ; brake:  0.0
```

Thanks for the help.

[image]

### Which car does this affect?

Honda Civic 2016

### Provide a route where the issue occurs

carla

### openpilot version

0.9.0

### Additional info

[image]

closed 2023-02-08 by didpurwanto

GregorKikelj, 2023-02-12:
@didpurwanto Any chance to learn how you resolved the issue?

didpurwanto, 2023-02-13:
@GregorKikelj you can resolve this issue by installing ``libzmq`` package to enable modeld module. First, ``apt purge libzmq* -y
``, then do this step-by-step
```
git clone https://github.com/zeromq/libzmq.git /libzmq
cd /libzmq
./autogen.sh
./configure
make
make install
ldconfig
```

GregorKikelj, 2023-02-13:
@adeebshihadeh Any idea why this is needed? Zmq is in cereal already...

zeroyang0809, 2023-02-20:
hello i meet the problem "openpilot unavailable waiting for controls to start"
this problem is familiar with your problem
do you know how to fix it ?
thx