commaai/openpilot #27256: 0.9.0 modeld not working and openpilot not accelarate

didpurwanto, 2023-02-08:
### Describe the bug

I have updated the openpilot to 0.9.0, but have several issues.

1. ``modeld`` not working
snapshot
```
create CUDAExecutionProvider. Please reference https://onnxruntime.ai/docs/reference/execution-providers/CUDA-ExecutionProvider.html#requirements to ensure all dependencies are met.
clocksd logcatd proclogd logmessaged modeld ui soundd locationd calibrationd torqued controlsd deleter laikad navd paramsd plannerd radard thermald statsd                                                                                                        
Onnx using  ['CPUExecutionProvider']
HERE ready to run onnx model ['input_imgs', 'big_input_imgs', 'desire', 'traffic_convention', 'features_buffer'] [[1, 12, 128, 256], [1, 12, 128, 256], [1, 100, 8], [1, 2], [1, 99, 128]]
---- HOHO ready to run onnx model ['input_imgs', 'big_input_imgs', 'desire', 'traffic_convention', 'features_buffer'] [[1, 12, 128, 256], [1, 12, 128, 256], [1, 100, 8], [1, 2], [1, 99, 128]]
-- GT  0  / SZ  393216  / ST  b''  /
Traceback (most recent call last):
  File "/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 75, in <module>
    run_loop(ort_session, tf8_input=("--use_tf8" in sys.argv))
  File "/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 48, in run_loop
    inputs.append(read(ts, (k=='input_img' and tf8_input)).reshape(shp).astype(itp))
  File "/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 21, in read
    assert(len(st) > 0)
AssertionError
```
Note that I have reinstalled libzmq package

2. openpilot not accelerate
```
frame:  engaged: False ; throttle:  0.0 ; steer(c/deg):  -0.0 0.0 ; brake:  0.0
```

Thanks for the help.

### OS Version

Ubuntu 20.04

### openpilot version or commit

0.9.0

### Additional info

[image]

PhilWallace, 2023-02-08:
I seemed to have solved this modeld problem by reinstalling the libzmp module, as indicated in https://github.com/commaai/openpilot/issues/23666. Maybe you can give it a try.

didpurwanto, 2023-02-09:
@PhilWallace do you install libzmq from source or ``apt-get install libzmq3-dev``? I have tried to uninstall and install it again with the apt-get but still not work. Thanks

didpurwanto, 2023-02-09:
@PhilWallace Thanks, I have solve it by install from source.

closed 2023-02-09 by didpurwanto

zeroyang0809, 2023-02-15:
hello could you share the  URL that  where i can find the openpilot  0.9.0 version? Thx

didpurwanto, 2023-02-15:
@zeroyang0809 you can go to openpilot github then choose the version.

zeroyang0809, 2023-02-15:
https://github.com/commaai/openpilot/tree/v0.9.0
is this right?
thx