commaai/openpilot #27257: The controls lagging problem in Carla Simulation

ferryman0723, 2023-02-08:
### Describe the bug

I'm running the latest Openpilot docker on Carla simulation.
After input 
```
# Terminal 1
./start_carla.sh
# Terminal 2 - Run openpilot and bridge in one Docker:
./start_openpilot_docker.sh
```
and input` ./bridge.py --high_quality --dual_camera` in the openpilot docker,the car cannot accelerate and keep driving in the lane after I input '1'.
[image]
After that I modify the code in selfdrive/controls/controlsd.py  like this
[image]
[image]

Although the car can accelerate now, it performs badly and slowly,unable to maintain lane driving and show the Controls Lagging problem,It seems that this problem existed before I modified the code.Has anyone met this problem and successfully solved it?Thanks a lot！

[image]
[image]
[image]
[image]

### OS Version

Ubuntu 20.04(docker)

fatsheep2020, 2023-02-08:
it can work on v0.9.0, not OP docker, after I tested

ferryman0723, 2023-02-10:
> it can work on v0.9.0, not OP docker, after I tested

how can I run the v0.9.0 in the docker as my PC is ubuntu18.04,have you met the same problem?thanks for reply!

ferryman0723, 2023-02-14:
some screenshots
[image]
[image]

jon-chuang, 2023-02-18:
What are your system specs? It's probably a performance issue if your GPU is weak.

Try running without `--high_quality` and report back. Thanks.

ferryman0723, 2023-02-19:
> 

It seems to have the same problem after running without --high_quality,still lagging problem.I use Intel(R) Xeon(R) Silver 4214 CPU + 2080TI,is this configuration enough to run op?thanks

ferryman0723, 2023-02-19:
> What are your system specs? It's probably a performance issue if your GPU is weak.
> 
> Try running without `--high_quality` and report back. Thanks.

It seems to have the same problem after running without --high_quality,still lagging problem.I use Intel(R) Xeon(R) Silver 4214 CPU + 2080TI,is this configuration enough to run op?thanks

zeroyang0809, 2023-02-20:
hello i meet the problem "openpilot unavailable waiting for controls to start" 
this problem is familiar with your problem 
do you know how to fix it ? 
thx

jon-chuang, 2023-02-20:
If you are building on master, it is due to https://github.com/commaai/openpilot/pull/27354

You need to add:
`Params().put_bool("ObdMultiplexingDisabled", True)`

zeroyang0809, 2023-02-20:
thanks for your reply 
where to add Params().put_bool("ObdMultiplexingDisabled", True)
thank you so much

jon-chuang, 2023-02-20:
Put it in the `peripheral_state_function` loop in `bridge.py`

jon-chuang, 2023-02-20:
Yes. Lmk if it works.

zeroyang0809, 2023-02-21:
thanks for your reply.
i can successfully launch the carla simulation on openpilot.
however when the car run for a few seconds,
it has this problem "take control immediately Vechicle Parameter Identification Failed". 
Do you know how to fix it? Thanks 
[image]

jon-chuang, 2023-02-21:
Not too sure. You could try to put the suggested line after pm.send(..) instead.

ferryman0723, 2023-02-23:
> What are your system specs? It's probably a performance issue if your GPU is weak.
> 
> Try running without `--high_quality` and report back. Thanks.

After the above steps, the latest op image in my PC has the following problems,I think it‘s still controls lagging problem.Does it means the GPU is not strong to run the op and carla?I use the 2080T
[image]
[image]

ferryman0723, 2023-03-19:
> it can work on v0.9.0, not OP docker, after I tested

hello,have you met the same problem？can you give me some measure ，thanks a lot!

ferryman0723, 2023-03-22:
> it can work on v0.9.0, not OP docker, after I tested

how can you run the v0.9.0  op docker,I can't find it ,can u give me some idea?thanks!

ferryman0723, 2023-03-22:
> it can work on v0.9.0, not OP docker, after I tested

how can you run the v0.9.0 op docker,I can't find it ,can u give me some idea?thanks! @fatsheep2020

ferryman0723, 2023-03-23:
> What are your system specs? It's probably a performance issue if your GPU is weak.
> 
> Try running without `--high_quality` and report back. Thanks.

@jon-chuang  I have the question that when I run the official docker,does it use the GPU?I havn't seen the op is using gpu

ferryman0723, 2023-04-01:
I resolved this problem after I changed my pc,I think it‘s Gpu server configuration issues and I recommend using op on personal computer.Anyway, thanks all those who replyed.You can use the docker from:(https://hub.docker.com/r/ferryman0723/openpilot).Now it can work.

closed 2023-04-01 by ferryman0723

CrazyMustard-404, 2024-01-09:
> thanks for you reply. is that like this?
> 
> [image]

Hello, could you please send me a copy of your bridge.py? I am currently using openpilot 0.8.9 and encountering the same issue as you. Additionally, my bridge.py differs from the latest official version. Thank you! Email: 415736313@qq.com

tk-menot, 2024-03-25:
@ferryman0723 Hi, considering that the latest version no longer supports Carla, I am using your Docker image. But I encountered some problems: (1) The UI interface of openpilot in the image is problematic, and I cannot see the surrounding environment. (2) I hope to set a destination and have the openpilot drive towards that point. Do you have any methods? Looking forward to and thank you very much for your answer!