commaai/openpilot #27282: Vehicle Stops After Vehicle Parameter Identification Failed Alert

didpurwanto, 2023-02-10:
### Describe the bug

I just updated to 0.9.0 version in docker version and encounter some problems. The first matter, which has been resolved by installing libzmq package to enable modeld module. The second issue that has not resolved yet is the vehicle start to decelerate and finally stop after Vehicle Parameter Identification Failed alert.

Snapshot of ``bridge.py``
[image]

Thanks for the help.

### OS Version

Ubuntu 20.04

### openpilot version or commit

0.9.0

### Additional info

[image]
[image]
[image]
[image]
[image]

PhilWallace, 2023-02-10:
Comment the identification failed alert code in _controlsd.py_ worked for me.

didpurwanto, 2023-02-14:
@GregorKikelj Do you face this problem too?

didpurwanto, 2023-02-14:
@PhilWallace Can you give more detail information?

GregorKikelj, 2023-02-14:
@didpurwanto I can't get UI to show the camera feed yet... It segfaults with dual camera mode but should finally have some time today to investigate more

PhilWallace, 2023-02-16:
> @PhilWallace Can you give more detail information?

I commented the following codes in selfdrive/controls/controlsd.py, should be line 374~375:
    #if not self.sm['liveParameters'].valid:
    #  self.events.add(EventName.vehicleModelInvalid)

didpurwanto, 2023-02-17:
Thansk @PhilWallace . Indeed it will work but will affect the driving quality.

PR #28685 (2023-06-25): sim: fix paramsd error
closes #27282
closed 2023-06-25 by adeebshihadeh