commaai/openpilot #27284: High quality simulator entry point

adeebshihadeh, 2023-02-10:
Clone openpilot, run the setup scripts, then run a single command to get dropped into the openpilot UI with it driving in the simulator.

It can be CARLA, but it's huge and requires way more HW than openpilot itself does. [SuperTuxKart](https://supertuxkart.net/) might be nice: apt-installable, small-ish, and modest HW requirements.

MoreTore, 2023-02-13:
I took a swing at MetaDrive. I can only get about 10 FPS on wls2 when copying the panda3d screenshot to vpc. Maybe I will try running it natively. 

Added this function to metadrive lib and converted the screenshot to YUV format and reused the carla bridge code.

```
def get_screenshot(self):
    if self.engine.episode_step <= 1:
        self.engine.graphicsEngine.renderFrame()
    origin_img = self.cam.node().getDisplayRegion(0).getScreenshot()
    return origin_img
    
```
   

 Metadrive is missing imu sensor inputs so that would need to be added eventually. 
 Metadrive also has a few compatibility issues with mac.

MoreTore, 2023-02-14:
Here is a little video. https://youtu.be/P-ti8c1esiY

adeebshihadeh, 2023-02-14:
Cool! No accelerometer/gyro should be fine, with changes to the localizer.

pfeiferj, 2023-02-15:
i found out there's a fork of torcs that can be controlled with CAN messages which could be interesting: https://github.com/epozzobon/torcs-1.3.7

MoreTore, 2023-02-16:
So I ended up getting some pretty stable performance with metadrive. However I am having other issues with getting the model to run reliably.  I am getting an assert error in onnx_runner.py. It works about 10% of the time.

```
def read(sz, tf8=False):
  dd = [] # data
  gt = 0 # got
  szof = 1 if tf8 else 4 # size of
  print("reading %d bytes" % (sz * szof), file=sys.stderr)
  while gt < sz * szof: # read until we have enough
    #print("here")
     st = os.read(0, sz * szof - gt) # we fail here 90% of the time reading stdout?? dup2(pipeout[1], 1); ??
    #assert(len(st) > 0) # make sure we got something
    dd.append(st) # append
    gt += len(st) # update
  r = np.frombuffer(b''.join(dd), dtype=np.uint8 if tf8 else np.float32)
  if tf8:
    r = r / 255.
  return r
```

MoreTore, 2023-02-16:
What are all the detailed requirements for this PR bounty? I basically have it working. Now it just needs polish.

jon-chuang, 2023-02-17:
@MoreTore could you post your updated results in YouTube again? I think we would optimize for better performance on lower spec device. 

I ensured that we are getting a similar field of view to the comma device cameras and that we are running at the intended resolution 1928 X 1208.

I should have unit tests and PR by EOD, e2e tests in https://github.com/commaai/openpilot/issues/26215 by tmr.

MoreTore, 2023-02-17:
Can't really get high resolution on low spe k device with the current game engine. We need to copy the Vram to the IPC, do a conversion, then put it back to do invfrence and display. This takes a long time and there is no way around it with the current game engine API. I am using a rtx3070ti with 12700H.

MoreTore, 2023-02-18:
@jon-chuang I just tested your branch and on my PC I only get 5 fps?

MoreTore, 2023-02-19:
>  let me know how it goes.
> 

Seems to work at 3 tpf. The bridge terminal control is really bad. Probably shouldn't use it for metadrrice

jon-chuang, 2023-02-19:
> The bridge terminal control is really bad. 

Could you elaborate?

MoreTore, 2023-02-20:
Make it like this https://youtu.be/ic1IKOWRzwQ

jon-chuang, 2023-02-20:
Yeah, if you're talking about the manual controls, I fully agree. You can't hold W while steering with A/D.

MoreTore, 2023-02-21:
Use metadrive for the inputs. It's already configured.

MoreTore, 2023-02-22:
Anyways I made this branch to modify metadrive in openpilot easier. Give it it a try. https://github.com/MoreTore/openpilot/tree/meta

MoreTore, 2023-02-22:
There is another simulator i was looking into. It shoul be avail on linux in a few months. Its high fidelity but it has shared gpu memory built in. I am talking about beamnd drive. https://beamng.tech/
Here is a sample script to get camera streams.
[stream.zip](https://github.com/commaai/openpilot/files/10808460/stream.zip)

MoreTore, 2023-04-13:
@adeebshihadeh Care to try this? I cleaned it up a bit.
https://github.com/commaai/openpilot/pull/27892

closed 2023-09-29 by adeebshihadeh