commaai/openpilot #27328: UI crash during hard stop

zunichky, 2023-02-14:
### Describe the bug

The OP interface crashed briefly, the black screen with the OP logo popped up, then the UI came back up. It did not seem like a full system reboot as the loading bar was never seen, and it was pretty quick to recover.  Looking at the footage, nothing stopped recording/logging. 
I was driving without OP engaged and almost missed my turn. I slammed on my brakes pretty hard and just about at the end of the braking, I noticed the screen went to the comma logo. 
ca4de5b12321bd98|2023-02-09--19-39-09--1 is when this happened. Right before I started my right hand turn onto the entrance ramp. 

I do not have prime and wasn't able to upload the logs/files in the 3 day window, but have them downloaded locally. Looks like the rlog got uploaded though. 

### Provide a route where the issue occurs

ca4de5b12321bd98|2023-02-09--19-39-09

### openpilot version

master-ci

### Additional info

Updated master-ci the morning of 2/9, did not catch the commit hash.

jaderaejackson, 2023-02-14:
Yes Jade.

deanlee, 2023-12-19:
@adeebshihadeh @sshane  I've encountered a few times where the UI gets kicked off by the watchdog and restarted.
it's caused by the vipc client can still connect to the ipc path even if the camerad is not running. This causes vipcClient::getavailableStreams to crash, https://github.com/commaai/cereal/pull/562 can fix it.
I'm not sure, but maybe this is the same issue.

adeebshihadeh, 2025-10-18:
Fixed with a beautiful red diff https://github.com/commaai/agnos-builder/commit/689537765d17b3132add1c5cab0f2356558af9d7!

closed 2025-10-18 by adeebshihadeh