commaai/openpilot #27358: `supercombo.onnx` causes segmentation fault in onnxruntime==1.14.0

jon-chuang, 2023-02-16:
### Describe the bug

```
python3 onnx_runner.py "../models/supercombo.onnx"
['onnx_runner.py', '../models/supercombo.onnx']
Onnx available providers:  ['CPUExecutionProvider']
Onnx selected provider:  ['CPUExecutionProvider']
Segmentation fault (core dumped)
```

navmodel.onnx runs fine
```
 python3 onnx_runner.py "../models/navmodel.onnx"
['onnx_runner.py', '../models/navmodel.onnx']
Onnx available providers:  ['CPUExecutionProvider']
Onnx selected provider:  ['CPUExecutionProvider']
Onnx using  ['CPUExecutionProvider']
```

Env:
```
pip show onnxruntime
Name: onnxruntime
Version: 1.14.0
Summary: ONNX Runtime is a runtime accelerator for Machine Learning models
Home-page: https://onnxruntime.ai
Author: Microsoft Corporation
Author-email: onnxruntime@microsoft.com
License: MIT License
Location: /home/jonch/.cache/pypoetry/virtualenvs/openpilot-P9NmkXAX-py3.8/lib/python3.8/site-packages
Requires: coloredlogs, flatbuffers, numpy, packaging, protobuf, sympy
Required-by:
```

Openpilot is on main

### Provide a route where the issue occurs

NUL

### openpilot version

d4c5b8d6f

jon-chuang, 2023-02-16:
Does not segfault on [1.13.1](https://github.com/microsoft/onnxruntime/releases/tag/v1.13.1)

jon-chuang, 2023-02-16:
Maybe it's an onnxruntime bug.

YassineYousfi, 2023-02-21:
The current supported onnxruntime version is 1.12.1 
https://github.com/commaai/openpilot/blob/master/poetry.lock#L2557

We currently only support that (1.13.1 might also work) 
Why do you need onnxruntime==1.14.0? We can update the models to work with 1.14, but that requires we resolve some breaking changes in onnx opsets.
https://onnxruntime.ai/docs/reference/compatibility.html#onnx-opset-support

Switching to discussion as this is expected behavior.