commaai/openpilot #27361: Fingerprinting takes too long. tmux errors

MoreTore, 2023-02-16:
### Describe the bug

Finger printing has been slow when using 9.0 for months. I thought it was just my car but it seems to work fine on my C2 with 0.8.13. I have been putting off reporting this issue. Here is the tmux output after deleting the CarVin and CarParams.

iso-tp query bad response: (2017, None) - 0x7f2231                                                                                                                                                                                           iso-tp query bad response: (1988, None) - 0x7f2231                                                                                                                                                                                                                                                                                                                                                    iso-tp query bad response: (2016, None) - 0x7f2231                                                                                                                                                                                           iso-tp query bad response: (2017, None) - 0x7f2231                                                                                                                                                                                           iso-tp query bad response: (1799, None) - 0x7f1012                                                                                                                                                                                           iso-tp query bad response: (1856, None) - 0x7f1012 

...

iso-tp query timeout while receiving data                                                                                                                                                                                                    Error processing UDS response: (1882, None)                                                                                                                                                                                                  Traceback (most recent call last):                                                                                                                                                                                                             File "/data/openpilot/selfdrive/car/isotp_parallel_query.py", line 108, in get_data                                                                                                                                                            dat, updated = msg.recv()                                                                                                                                                                                                                  File "/data/openpilot/panda/python/uds.py", line 445, in recv                                                                                                                                                                                  self._isotp_rx_next(msg)                                                                                                                                                                                                                   File "/data/openpilot/panda/python/uds.py", line 488, in _isotp_rx_next                                                                                                                                                                        assert self.rx_idx & 0xF == rx_data[0] & 0xF, "isotp - rx: invalid consecutive frame index"                                                                                                                                              AssertionError: isotp - rx: invalid consecutive frame index                                                                                                                                                                                  Error processing UDS response: (1882, None)                                                                                                                                                                                                  Traceback (most recent call last):                                                                                                                                                                                                             File "/data/openpilot/selfdrive/car/isotp_parallel_query.py", line 108, in get_data                                                                                                                                                            dat, updated = msg.recv()                                                                                                                                                                                                                  File "/data/openpilot/panda/python/uds.py", line 445, in recv                                                                                                                                                                                  self._isotp_rx_next(msg)                                                                                                                                                                                                                   File "/data/openpilot/panda/python/uds.py", line 488, in _isotp_rx_next                                                                                                                                                                        assert self.rx_idx & 0xF == rx_data[0] & 0xF, "isotp - rx: invalid consecutive frame index"                                                                                                                                              AssertionError: isotp - rx: invalid consecutive frame index                                                                                                                                                                                  iso-tp query bad response: (2016, None) - 0x7f2231                                                                                                                                                                                           iso-tp query bad response: (2017, None) - 0x7f2231 

...
                                                                                                                                                                                                                                                                                                                                                                                      iso-tp query bad response: (2016, None) - 0x7f2231                                                                                                                                                                                           iso-tp query bad response: (2017, None) - 0x7f2231                                                                                                                                                                                           iso-tp query bad response: (2017, None) - 0x7f2231                                                                                                                                                                                           iso-tp query bad response: (2016, None) - 0x7f2231                                                                                                                                                                                           iso-tp query bad response: (2016, None) - 0x7f2231                                                                                                                                                                                           iso-tp query bad response: (2017, None) - 0x7f2231                                                                                                                                                                                           iso-tp query bad response: (1840, None) - 0x7f2231                                                                                                                                                                                                  [... cut ...]                                                                               selfdrive/boardd/boardd.cc: finished FW query at OBD port                                                                                                                                                                                    selfdrive/boardd/boardd.cc: waiting for params to set safety model 

### Provide a route where the issue occurs

164080f7933651c4|2023-02-16--08-52-15--0

### openpilot version

0.9.0.x

sshane, 2023-02-16:
Going to re-open until I check the timeouts to make sure they're working properly. @MoreTore how long does it take to complete FW querying for you currently?

sshane, 2023-02-17:
Going to re-open until I check the timeouts to make sure they're working properly. @MoreTore Can you also connect your device to WiFi to upload logs? https://github.com/commaai/openpilot/pull/25897 may be related

MoreTore, 2023-02-17:
So the can gateway in not on the camera, its on the BCM. Not sure if that matters since its using the OBD port. But anyways 164080f7933651c4|2023-02-16--08-52-15 should be uploaded.

adeebshihadeh, 2023-02-17:
@sshane can we close this?

MoreTore, 2023-02-17:
yea this is my fault

closed 2023-02-17 by MoreTore