commaai/openpilot #27396: Frequent "Turn Exceeds Steering Limit- Take control" warnings on curves the car can handle

jettoblack, 2023-02-19:
### Describe the bug

Getting frequent "Turn Exceeds Steering Limit- Take control" warnings on curves that the car is able to handle with 0 driver intervention. At no point did I have to touch the wheel or brake in order to successfully navigate these curves. If an actual limit were being hit, the car should have required some driver intervention to stay in lane.

Examples:

48c21cb1105d4d1c|2023-02-19--17-10-19--7 at 17:17:42

48c21cb1105d4d1c|2023-02-19--16-05-48--3 at 16:09:36, 16:10:15, 16:10:43

48c21cb1105d4d1c|2023-02-19--15-59-06--2 at 16:01:24, 16:02:47, 16:03:13, 16:03:52

### Which car does this affect?

Bolt EUV 2022

### Provide a route where the issue occurs

48c21cb1105d4d1c|2023-02-19--17-10-19--7

### openpilot version

8d9f99b commaai/master

Verylukyguy, 2023-02-20:
This also happened frequently on the 2018 GMC Acadia.

I get the warning while in a curve and it usually does not require driver intervention, most times, it turns more while the warning is displayed.

I'm not sure if tuning has anything to do with it, but when I got a custom FeedForward, this issue was greatly reduced.

sshane, 2023-08-08:
@jettoblack does this still happen with the latest tuning?

jettoblack, 2023-08-17:
> @jettoblack does this still happen?

I haven't used stock OP in a while, as I find stock completely unusable without an independent toggle for lateral control (aka "MADS").  But this issue still happens in other forks based on the latest OP, so AFAIK, it still happens in OP.

@twilsonco came up with the idea of using steerLimitTimer and only emitting this warning if the understeer / max torque condition persists for longer than that. I lost his commit in his sea of other branches, but I have the equivalent change in my branch here:
https://github.com/jettoblack/openpilot-1/commit/b3a910d887bbc5c6921cf3b078df2bbfb2984d1a

This reduces the warning incidence by about 90%+.  Now it only happens on really sharp curves, like cloverleafs. (But, even then, the car completes the curve just fine without any driver intervention; i.e. it is NOT actually torque limited.) Without this change, I get this warning constantly even on slight curves.