commaai/openpilot #27399: Openpilot unavailable Waiting for controls to start

zeroyang0809, 2023-02-20:
### Describe the bug

when i run the car simulation on openpilot, i meet this problem.
i run the openpilot on local(instead of docker).
there is no any error on there terminals.

[image]

### Provide a route where the issue occurs

no Dongle ID 

### openpilot version

0.9.0

zeroyang0809, 2023-02-20:
i find that my Dongle ID is none. is that normal? thanks

[image]

jon-chuang, 2023-02-20:
See also: https://github.com/commaai/openpilot/issues/27257#issuecomment-1436319753

zeroyang0809, 2023-02-21:
thanks for your reply.
i can successfully launch the carla simulation on openpilot.
however when the car run for a few seconds,
it has this problem "take control immediately Vechicle Parameter Identification Failed".
Do you know how to fix it? Thanks
[image]

joedursun, 2023-02-22:
Getting this same error on my 2020 Subaru Outback with a brand new Comma 3. I'm new to this so not sure if it's a an installation issue (I've double checked that the harness and cables are fully plugged in) or a bug in OpenPilot.

MoreTore, 2023-02-22:
> thanks for your reply. i can successfully launch the carla simulation on openpilot. however when the car run for a few seconds, it has this problem "take control immediately Vechicle Parameter Identification Failed". Do you know how to fix it? Thanks [image]

this is likely the fake gps or imu data not updating correctly. You can just disable it and add this to controlsd:

if SIMULATION:
  ignore += ['driverCameraState', 'managerState','liveParameters', 'liveTorqueParameters', 'liveLocationKalman']

closed 2023-03-13 by adeebshihadeh

davidx, 2024-02-29:
I just experienced this too on c3x toyota prius 2017