commaai/openpilot #27451: Chevy Bolt EUV Premier 2023 - Autopilot fails to restart with no Comma error message after stop and go city traffic

DerrickBarra, 2023-02-25:
### Describe the bug

**Vehicle**
Chevy Bolt EUV 2023 Premier

**Device**
Comma 3 with Bolt EUV 2023 Harness

**Autopliot Version**
Autopilot Master Branch - February 21st 2023 release

**Description**
Autopilot fails to restart after allowing it to previously come to a complete stop in traffic.

**To Replicate**
1. Allow Autopilot to slow the Chevy Bolt down at a traffic light behind another car and come to a complete stop.
2. Allow enough time to pass so that Autopilot is not longer able to control the Bolt and will require it to be disabled and manual control by the driver to take over.
3. When traffic resumes, disengage autopilot using the gas pedal to resume standard vehicle control and follow traffic
4. After a few seconds, attempt to re-engage Autopilot.
5. If the bug occurred, you won't see a message appear on the Comma 3. However Autopilot will fail to start with no feedback from the Comma 3 or the driver info screen on the Bolt.

**Notes**
- In one instance when this error occured, I believe I saw a message on the Chevy Bolt's driver info screen that said "Adaptive Cruise Control Unavailable". But usually I'm not looking at the driver info screen when this error occurs so it can be hard to catch it.

**Example Route**
[Recorded Route Link](https://connect.comma.ai/3b68b660a77fd80e/1677281083389/1677282764840)
Around 18:30'ish (around 6:30PM EST) on this route is when I noticed the issue.

### Which car does this affect?

Chevy Bolt EUV Premier 2023

### Provide a route where the issue occurs

3b68b660a77fd80e|2023-02-24--18-24-43--6

### openpilot version

f381b694013ef662c5d7638fe7b706315ce9547d

nelsonjchen, 2023-02-25:
Hey uh, the thing is called openpilot. Just wanted to say it's a bit awkward to keep seeing a competitor's name.

DerrickBarra, 2023-02-25:
> Contributor

Good catch! I've modified the post accordingly.

sshane, 2023-03-02:
Unfortunately the camera does not publish anything on CAN telling us that it's faulted. Do you see any message on the dash when this happens? How long after you disengaged at a stop did this happen?

sshane, 2023-03-25:
@DerrickBarra Does this still happen for you?

sshane, 2023-05-10:
Any updates @DerrickBarra? Closing if not.

sshane, 2023-05-30:
I'll try to reproduce here, it's very possible that the camera does not like seeing duplicate counters from the buttons.

[image]

adeebshihadeh, 2024-01-05:
Is this still an issue?

closed 2024-02-16 by adeebshihadeh