commaai/openpilot #27461: 2023 Hyundai Tucson Hybrid identification

ishfaaq, 2023-02-28:
### Describe the bug

The 2023 Tucson Hybrid is identified as the ICE model because the camera and radar modules have the same Id. It works perfect once modified.

### Which car does this affect?

Hyundai Tucson Hybrid 2023

### Provide a route where the issue occurs

92c61194164c3e2e|2023-02-26--10-01-51

### openpilot version

0.9.1

### Additional info

Please let me know if you need additional info about the car

sshane, 2023-03-01:
Hi @ishfaaq, it would be helpful if you could run a command over ssh. Are you on Discord? That might be quicker to communicate.

ishfaaq, 2023-03-02:
I am, I posted this in the hkg channel with the username ishj. I got it to run for now by modifying values.py, but I'll happily test things.

mathiasparks, 2023-04-13:
Hi @ishfaaq & @sshane - Looking for your post on the hkg discord, but having some trouble finding it. I attempted to setup the Comma 3 in my 2023 Hyundai Tucson Hybrid yesterday evening and got the "Dashcam Mode Car Unrecognized" message. Any chance either of you could provide a link to the resolution for this? Much appreciated :)

sshane, 2023-04-13:
@mathiasparks can you provide your dongle id?

not-vanished, 2023-04-19:
> @mathiasparks can you provide your dongle id?

Hey Shane, I'm a long time follower of my your branch on my C2, and super happy for you for being in the SD area. I grew up in Denton County as well! Hope your week is going well, man!

Could you help me as to locating the dongle ID, is that's physically looking at the hardware? I'm looking to get my 2023 Hyundai Tucson Hybrid running as well.

mathiasparks, 2023-04-19:
Hey @sshane - sorry I've been out of town and am now back home. My dongle id is a1f50efb97003e30.

sshane, 2023-04-25:
@not-vanished sorry for the late reply, yes that will be in the Device settings! Do you already have it set up?

sshane, 2023-04-25:
@mathiasparks @not-vanished are you the same person?

not-vanished, 2023-04-25:
> @mathiasparks @not-vanished are you the same person?

I am not the same person. Heh.

sshane, 2023-04-26:
@not-vanished any updates?

not-vanished, 2023-04-26:
@sshane Hey Shane, sorry for the delay. I had to fly out of town for an emergency. Let me get this info to you by the weekend.

sshane, 2023-04-26:
No worries! Whenever you get the chance!

sshane, 2023-04-26:
@mathiasparks adding your car in: https://github.com/commaai/openpilot/pull/28036

Doesn't seem to conflict with existing FW versions for the non-hybrid so should be safe to merge.

sshane, 2023-04-26:
@mathiasparks adding your car in: https://github.com/commaai/openpilot/pull/28036

Doesn't seem to conflict with existing FW versions for the non-hybrid so should be safe to merge. There's been a problem where some Tucson Hybrid and ICE share identical FW versions, but everything looks fine with your car.

Did you use the Hyundai N harness?

mathiasparks, 2023-04-28:
Thank you @sshane! Much appreciated. How do I get that build on my Comma 3? 

Regarding harness, are you referring to the Red Panda?

not-vanished, 2023-04-28:
> No worries! Whenever you get the chance!

Hey @sshane, I messaged you directly on Discord as Kenichi#4323 with my Dongle ID and VIN. Thanks again for your help!

not-vanished, 2023-04-30:
@sshane - @mathiasparks is actually a friend of mine! We didn’t realize we were replying to this thread together. The car is now working! Thank you so much for your help and continued contributions to the OpenPilot community.

sshane, 2023-05-01:
Looks like you're both supported now! Let me know if there's any other issues or feedback you have!

closed 2023-05-01 by sshane

sohaib0004, 2023-10-29:
@sshane Hey I am having same issue with my 2024 tucsan hybrid, I have sent you friends request on discord. Hope you can help me out too

ishfaaq, 2023-10-30:
@sohaib0004 You will need to replace the firmware(s) in the Hybrid section with the ones from the ICE section. You can just switch them around. Openpilot is currently thinking that your car is an ICE model.

sohaib0004, 2023-10-30:
@ishfaaq can you give me quick call 3464344362. I am not familiar with github / discord so its hard for me to go inside any of the comma3 files. If you can help me with that i can change it.