commaai/openpilot #27465: Lexus RX 2022 - “dashcam mode, car unsupported” error

adub8923, 2023-03-01:
### Describe the bug

Installed Comma 3 in my Lexus RX 350 2022 and my Comma 3 read “Dashcam mode, car unsupported”.

I fingerprinted the car and figured out how to fix this. I don’t know how to add a Pull Request and was hoping this fix could be added to future updates.

in values.py folder
  under category CAR.LEXUS_RX_TSS2
  under subcategory Ecu.engine
  ADD this line b’\x01896630ED5000\x00\x00\x00\x00’,

Now Comma 3 works in the car!

### Which car does this affect?

Lexus RX 350 (2022)

### Provide a route where the issue occurs

Fixed it

### openpilot version

0.9.1

nelsonjchen, 2023-03-01:
Routes/Segments IDs are like this: https://github.com/commaai/openpilot/wiki/FAQ#how-do-i-report-a-bug

sshane, 2023-03-01:
Can you provide your dongle id?

sshane, 2023-03-25:
@adub8923 any updates? We just need your dongle id to determine your FW versions to add to openpilot. It will be in Device settings

adub8923, 2023-03-25:
Hi, yes my dongle id is be55f95d3f668fcc

Thank you.

sshane, 2023-04-26:
Apologies for the late response, I don't see you are driving anymore, are you around still?

sshane, 2023-04-26:
I merged your fingerprint in for when you're ready to try it again! https://github.com/commaai/openpilot/pull/28035

closed 2023-04-26 by sshane

jaylinnyc, 2023-05-07:
I have the same issue with my comma 3, but 2021 Lexus Rx 350.   Will this fix work for my vehicle?