commaai/openpilot #27474: Can bus error on remote start

Miniz199922, 2023-03-01:
### Describe the bug

If I remote start my jeep I get a can bus error. If I start normally it works fine. 

### Provide a route where the issue occurs

5c77528ab6bf019a|2023-03-01--17-34-37--0

### openpilot version

0.9.1

sshane, 2023-03-01:
Can you provide a route on stock openpilot? I'm not able to parse the logs. If it's not easy, no worries I can pull the fork's log definitions. Is this CAN error persistent, or does it go away once you fully turn your vehicle on?

Miniz199922, 2023-03-02:
5c77528ab6bf019a|2023-03-01--22-05-29--0

Miniz199922, 2023-03-02:
It goes away when turn car on but openpilot doesn't auto steer.

sshane, 2023-03-02:
Can you get a route where you remote start the vehicle, then get in and put it in drive to start moving? I'd like to see what fault code the EPS reports when it finally starts sending, since you say you don't get steering that way.

Miniz199922, 2023-03-02:
It engages but no torque. I'll do it right now.

Miniz199922, 2023-03-02:
5c77528ab6bf019a|2023-03-01--22-25-37--0 this is the fault. Didn't let me even enable acc it started flashing red.

sshane, 2023-03-02:
That's interesting, it doesn't report a fault until openpilot starts sending torque. Can you make sure to get routes on master next time? But this route is on a fork, and you are spoofing the speed to steer at lower speeds, so it's hard for me to rule those out as reasons for the fault.

Can you get a route on master where you:
- remote start the car
- get in and drive
- try to engage above the higher minimum steering speed (~40 mph)

Also would be interesting to see if the stock LKAS system would fault in this instance. For that, it would be a separate test where you keep openpilot disabled in the settings and do the same actions above, but you'd need to find a highway so the stock system would try to apply torque.

Miniz199922, 2023-03-02:
5c77528ab6bf019a|2023-03-01--16-35-41--0 this one is stock happened on my way home from work. Above 40 setting acc no steer.

Miniz199922, 2023-03-04:
Currently the wp is not always powered only when. Push button start happens. Should the wp always be powered? I'm wondering if that's what causes the can issue on remote start. Not sure if the wp should always be powered.

Miniz199922, 2023-03-04:
I think I figured it out. I'll close for now and reopen if problem comes back. Ethernet cable wasn't fully seated.

Miniz199922, 2023-03-04:
5c77528ab6bf019a|2023-03-01--16-35-41--5

Miniz199922, 2023-03-04:
Fixed. Panda always needs power not switched power on Chrysler. This eliminates the can error.

Miniz199922, 2023-03-04:
Guess it's not fixed. 10 minutes ago started truck got lkas error. Restarted vehicle no lkas error. Maybe there needs to be a delay somewhere?

adeebshihadeh, 2023-03-07:
I see you're running a fork with some extra hardware. We can't look into issues from forks. If you can reproduce this with stock openpilot and the extra hardware completely removed, please open a new issue.

closed 2023-03-07 by adeebshihadeh