commaai/openpilot #27479: race condition when controls allowed times out near an engagement

sshane, 2023-03-02:
Basically, get panda's `controlsAllowed` to rise but make it a noEntry to openpilot. Then around 2 seconds later, engage.

For example in a button enable car: press set with seatbelt off, then put seatbelt back on and press set around 2 seconds later.

d4958b8ee5ef5cb3|2023-03-01--09-56-23--1

adeebshihadeh, 2023-03-09:
Another case
> cameron
>   [24 minutes ago](https://commadotai.slack.com/archives/C2T3FFDA6/p1678331460132779)
> cruise fault in the Bolt https://connect.comma.ai/6e997623cb7cdaa6/1678246341646/1678246375853

sshane, 2023-03-10:
Another case in https://github.com/commaai/openpilot/issues/26016

> Got another straight line red error a couple days ago: 0def4a390f6fe5c0|2023-03-09--10-36-17--93 I posted in tuning, is there a good place to be able to get better at describing an error? The car isn't always in a place where I can SSH into it but I am connected to connect to pull logs and such.

sshane, 2023-03-31:
Another *suspected* case 147613502316e718|2023-03-30--22-51-22--5

sshane, 2023-06-19:
Another case from support a27fe4e91d6e73c8|2023-06-09--13-02-14