commaai/openpilot #27514: [C3: No boot up: below 35°F]

r8j3, 2023-03-07:
### Describe the bug

Model: Comma 3 (late 2022)
Installed: about a month back
Issue: No boot/screen up below 35°F

As per request from @adeebshihadeh creating the issue.
https://discord.com/channels/469524606043160576/954493346250887168/1082432343198945340

37°F today at 4pm, C3 screen does not seem to come ON 😖. No boot up comma logo, no indication of life.
Had to drive around a bit, stop the car, pull C3 out, remove the wires for about 60 seconds or so, put it back, and start the car, and it comes on. Since I've observed this a few time, I even pre-heated the car (EV6) for more then 5 minutes.

This is probably the forth time in less than 10 days.

When temp is above 40+°F, no issues booting up.

### Provide a route where the issue occurs

12d8cdfbdb79be90

### openpilot version

0.9.1

adeebshihadeh, 2023-03-07:
Thanks for reporting this. The first step is to figure out whether it's the display that doesn't init or the whole device doesn't boot. Here's a couple ways to tell:
* are there any drives on connect for the times where this happened?
* when this happens, is openpilot or the stock HDA system working when you engage?

r8j3, 2023-03-07:
> * are there any drives on connect for the times where this happened?
I could not find the route for today on useradmin
> * when this happens, is openpilot or the stock HDA system working when you engage?
I believe it is the stock HDA system that is working.
When this happened the second time, I reported on discord and I remember @geohot mentioning that the device should still work, just that the boot loader screen init doesn't. Based on that thought the last 2 times I double checked with the cruise button and sure looked like in these instances, the stock HDA was working.

lskov, 2023-03-14:
I'm having the same issue and have experienced this for some time now. 
For me if the temperature is below 40°F its boots and the system work well, but the display is not running.
I didn't have any issues with this last winter with 15°F so I would assume something might have changed in agnos.

adeebshihadeh, 2023-06-08:
We shipped a fix for this in the OS update that came with openpilot 0.9.2. Are either of you still experiencing this problem?

adeebshihadeh, 2023-06-25:
The new AGNOS should fix any software-related problems here. If you're still having issues, please open another issue.

closed 2023-06-25 by adeebshihadeh

lskov, 2023-10-25:
The issue with cold weather appears to still be present in the latest master.

In order to ensure this is addressed, it is recommended to open a new issue?

adeebshihadeh, 2023-10-25:
Yes, please open a new issue and make sure to include routes if possible.