commaai/openpilot #27519: Pacifica 2017 Hybrid : FCW disabled & needs service in dashboard

gboudreau, 2023-03-07:
### Describe the bug

Last evening, during a drive 918f274474111392|2023-03-06--19-29-07, at around 19h41 I think, the car started complaining about FCW being disabled and needing to be serviced.
The comma (running openpilot level branch) also complained that something was wrong and that I should restart the car to fix it.
So at the next red light, I stopped the car and restarted it. This is where this drive ends, and the next one starts: 918f274474111392|2023-03-06--19-41-55.
At this point, the comma showed an unrecognized car (in useradmin.comma.ai, platform = mock for this drive).
When I arrived home, I stopped the car (plugged it in) and tried again. Same thing. This was around 19h51.
More than 20 mins later, I tried again to start the car. The warnings are now gone. 918f274474111392|2023-03-06--20-16-30

I'm worried that the comma's panda was somehow responsible for the car's FCW system being disabled...
Is there anything I could try to debug this further?

### Which car does this affect?

Chrysler Pacifica 2017 Hybrid

### Provide a route where the issue occurs

918f274474111392|2023-03-06--19-29-07

### openpilot version

0.9.1

### Additional info

I installed my new comma three a few days ago.
Everything went right during installation, apart from the comma showing "no panda connected" during installation; I unplugged the panda, re-connected it, and it was fine after that.
I drove for a few days, some highway, some city driving. Everything was fine.

gboudreau, 2023-03-18:
This happened again today.
I was near the end of a 2h45m trip (260 km). Chill mode was engaged, and I think it disengaged when I pressed a pedal (not sure which one). I added a bookmark to the trip a few seconds later.
The route : 918f274474111392|2023-03-18--11-10-32
The problem occurred around 13:47:03

sshane, 2023-03-28:
Interesting, the panda went into a bus off state when it faulted. Does this still happen? Has this happened when the comma three wasn't plugged in, or in dashcam mode?

[image]

gboudreau, 2023-03-28:
It only happened those two times, while I was driving and using the comma for assisted driving.
Could this somehow be caused by the panda cable not being properly plugged in? I remember during installation, after I plugged the cable the first time, the comma was complaining about not being connected to the car, even though the cables were all connected. I had to unplug the panda, and replug it (the same way) and then the comma said all was ok. But maybe somehow the panda cable is not completely connected or something, and when the car is moving, the cable could disconnect for a fraction of a second..?

sshane, 2023-03-28:
The panda is built into the comma three. Do you mean the relay box (what the three's USB C plugs into)? Yeah, it wouldn't hurt to try another USB C cable for an extended period time to make sure it works.

gboudreau, 2023-03-28:
Yes, I was talking about the relay box, and the cables it uses to connect to the car (not the three). That's the cables I had to disconnected and re-connect, during initial installation, for the three to "see" the car.
But if you think the USB-C cable is the most-likely culprit, I can try another.

sshane, 2023-03-28:
It could be either, we can start with checking if the USB C is the culprit though. Does this happen often?

gboudreau, 2023-03-29:
> Does this happen often?

Sadly (or happily!) no. It occurred only two times since I installed 4 weeks ago.
Still, I enable chill mode as often as I can on my morning and evening commute, even if I'm only traveling small roads, to try to reproduce the issue...

familycio, 2023-05-07:
Hello there, 

I had this "ACC/FCW Unavailable Service Required" issue with my 2020 Pacifica Hybrid, running new Comma 3 and standard Openpilot code. On a April 28 morning drive (20 min, around 8:20am), my Pacifica showed only one lane detected in ACC, but Comma 3 screen has both lane (and red curb) detected. On the same day return trip (around 1pm), my Pacifica showed the "ACC/FCW Unavailable Service Required" and OP was completely dead. 

My Pacifica has been in "ACC/FCW Unavailable Service Required" state ever since - I disconnect my Pacifica car battery over night trying to reset ACC/FCW, but still persist. I use the Comma 3 with the same OP version on my Nissan cars (Rogue and Leaf) with Nissan harness, and OP works fine. Can anyone help me to understand what's the problem in Comma3 with Pacifica hybrid 2020 before my visit to the dealership -  It costs about $500 to visit the Chrysler dealership and diagnose the ACC/FCW error. I don't want to repeatedly visit dealer if the issue is not fixable from their side.  Could it be the issue of failure Chrysler harness? 

More background, my Comma 2 stopped working on April 20, 2023 after running for like 3 years. I then purchase Comma 3 to replace it.  

Strangely I didn't see my April 28 video in Comma connect yet. I will connect my Comma 3 overnight and see if it upload, The last successful trip video on Nissan Rogue video is here: f421e8c2744fed7f|2023-05-06--18-14-01--1

Thanks!

sshane, 2023-05-10:
@familycio were you able to find a route where this happens?

@gboudreau has this even happened since?

familycio, 2023-05-10:
Sorry, I left both Comma 2 and Comma 3 overnight, but my Comma connect didn't show any driving history of the date when only one lane was shown on the Pacifica 2020 hybrid. I will bring my Pacifica for service tomorrow and will try Comma 3 again after the service. I will update this thread. It seems I missed the window of uploading the drive video?

sshane, 2023-07-11:
Has this happened since?

familycio, 2023-07-11:
Not any more after dealer serviced the carSent from my iPhone

gboudreau, 2023-07-22:
> Has this happened since?

It happened to me at least twice since April, so maybe 4-5 times in total.
Every time, shutting down the car, and waiting approx. 5 minutes, fixed the issue.

sshane, 2023-08-04:
@gboudreau can you send us some recent examples (preferably with full logs uploaded, but not required)? Make sure you're running a recent version of openpilot as we've fixed some things with panda recently. We might have an idea what is causing this if you can send more examples. Thanks!

gboudreau, 2023-09-12:
So my wife told be she was able to reproduce easily: just engage the regular cruise control (not the adaptative one), and hit `Set/+`, and that triggers this problem.
I tried today, and at first, I thought it didn't occur, but later during the trip, I did notice the comma was reporting a fault.

I hit the `Set/+` around 16:01:37 in this trip : `918f274474111392|2023-09-12--15-57-43--3`
But I only noticed the fault on the comma later, so I'm not sure when the fault occurred exactly.
I know that when I was at `918f274474111392|2023-09-12--16-15-26--26`, it was for sure showing.

sshane, 2023-09-12:
It may be possible that's a different fault from the one the issue is about. That is a steering fault, likely caused by the fact that we keep the LKAS control bit high the entire time, even when not steering. You can see right when you turned your ignition off in `918f274474111392|2023-09-12--15-57-43`, the temp steer fault bit rose.

[image]

Can you try a branch?

gboudreau, 2023-09-12:
On @ 18:53:15, off @ 18:53:30 : `918f274474111392|2023-09-12--18-53-14--0`
On again @ 18:54:00 : `918f274474111392|2023-09-12--18-53-59--0`

When I have this problem normally, I need to turn off, wait approx 5 minutes, and then the fault goes away. I only waited 30s this time. Is this ok, or should I turn it back on after 18:59?

sshane, 2023-09-12:
I didn't see any perm faults this time, has it ever happened following these instructions?

To make it clear, it sounds like there's two faults here. Permanent steering faults when turning the car on, and that ACC/FCW warning, am I correct?

gboudreau, 2023-09-12:
ACC fault on the dash: yes. That is what my wife was telling me about. She tried to use the regular CC, and when she tapped `Set/+`, she got the initial error I reported, on the comma and the dashboard, and couldn't turn on any CC after that, until the car was reset. Every time she tried to use the CC, this happened, she tells me.  
So I tried to reproduce this today, and I didn't see the error on the dashboard indeed, only on the comma. So maybe what I had today was different.

I'll go do some tests in the streets around, using the current branch, and report my findings later tonight when I get back.

gboudreau, 2023-09-12:
OK, so I think I know how to reproduce :

0. Started the car here : `918f274474111392|2023-09-12--19-13-19--0`

1. Try using the regular CC: as soon as I start it (using the `Set/+` button, see `918f274474111392|2023-09-12--19-13-19--1` @ 19:14:44), it disables itself in less than a second. No errors on the comma, northe car's dashboard. I tried again at 19:16:24, 19:16:29, 19:16:40 & 19:16:44.

At this point I can use the comma driving by using the ACC. See @ 19:17:24 when I tried that.

2. (Stop then) Turn off the car.

3. Turn it back on (don't wait 5 minutes) `918f274474111392|2023-09-12--19-19-29--0`
The comma shows "LKAS fault" here, but nothing yet on the dashboard.

Trying to use the regular CC (eg. at 19:19:53) has the same behaviour as previously: it doesn't work (it disables itself almost instantly).

5. Try to engage the comma now, using the ACC (19:20:24). After about 2 seconds, the comma disengage, it shows `Cruise fault` (was `LKAS fault` before) on its screen, and the car's dashboard now shows the initial error: `ACC/FCW Unavailable Service Required`. Trying to use either CC or ACC now doesn't work. (And FCW is also disabled.)

So basically, I think trying to use the regular CC, turning off the car, turning it back on, and trying to engage the comma, results in the initially reported error. Maybe not for the same reasons (the situation is different), but the symptoms are (I think) the same: car's dashboard & comma show errors, and CC/ACC/FCW are all disabled, until I turn off the car for 5+ minutes.

gboudreau, 2023-09-13:
Shorter run where I reproduced the problem: `918f274474111392|2023-09-12--19-54-29--0`
- Activate regular CC: 19:55:46 (it disengaged by itself within a second); no errors on dashboard or comma
- Stop the car (19:55:59)
- Restart the car: `918f274474111392|2023-09-12--19-56-09--0`
- `LKAS fault` on comma, no errors on dashboard
- Activate comma (ACC) 19:56:39 - it disengaged itself within a second; no error on dashboard, still `LKAS fault` on comma
- Activate comma (ACC) 19:57:47 - it worked. I disabled it myself after a few seconds.
- Activate regular CC: 19:58:27 (it disengaged by itself within a second); no error on dashboard, still `LKAS fault` on comma
- Activate comma (ACC) 19:58:40 - it worked for approx 5 seconds, then disengaged by itself. `Cruise fault` on comma, and dashboard shows `ACC/FCW Unavailable`

Galantha, 2023-11-01:
Does anyone have a solution to quickly reset the car when this happens?

gboudreau, 2023-11-01:
I didn't try it, but I'm pretty sure disconnecting the 12v battery, the (+) terminal, would reset it pretty quickly. Not sure what tool one would need to keep handy, and what else would be reset by doing this...

sshane, 2024-03-20:
Can you retry the above tests and provide a route with uploaded logs? The route is too old to upload.

gboudreau, 2024-03-23:
@sshane Do you need me to reproduce using the latest version, and/or the https://installer.comma.ai/commaai/pachy-2017-test branch?

Of note: for the last tests I did, I clicked `preserve` for the routes in question.
I still see those routes in https://useradmin.comma.ai :

[image]

Those should still contain the needed logs, right?