commaai/openpilot #27539: Camera failed to start. Caused dashboard lights

MoreTore, 2023-03-09:
### Describe the bug

Started the car, comma 3 says shows Multiple publisher errors. Dcam, widecam, roadcam. Cameras start up a few seconds later and Openpilot works but the car had cruise control error lights. Cycling the ignition fixed the issue.

164080f7933651c4|2023-03-08--15-43-14--0

### Provide a route where the issue occurs

164080f7933651c4|2023-02-16--08-52-15--0

### openpilot version

0.9.2

adeebshihadeh, 2023-03-09:
We can't look into issues from forks. Report the issue to the fork maintainers instead. Feel free to open a new issue if you can reproduce this with stock openpilot.

closed 2023-03-09 by adeebshihadeh

MoreTore, 2023-03-09:
> We can't look into issues from forks. Report the issue to the fork maintainers instead. Feel free to open a new issue if you can reproduce this with stock openpilot.

Ok. This branch is based on master ci from a few weeks ago. I only made changes to add my car.

adeebshihadeh, 2023-03-09:
`MultiplePublishersError` is easily caused by not properly restarting openpilot, running multiple instances of a service, etc. Feel free to reopen if you have a route where this happens in a clean environment on a stock branch.