commaai/openpilot #27551: Radar read error on Genesis G70 2020

jmsether, 2023-03-10:
### Describe the bug

At 1:08AM on the provided route ID, the comma unit states radar read error. This shows up rarely and at very random moments On stock Open pilot but reliably shows up on forks(specifically sunny pilot). Currently unknown on how to reproduce reliably. I would go a week without the issue showing up then next week I might get it daily and not be able to use Open pilot for the rest of the day.

### Which car does this affect?

Genesis G70 2020

### Provide a route where the issue occurs

dc4472af9d9d3fc5|2023-03-10--00-54-55--13

### openpilot version

0.9.1

jmsether, 2023-03-17:
Accidentally clicked the close button. I have a couple more routes on stock that caused this.
dc4472af9d9d3fc5|2023-03-15--11-30-05--5
At 11:35 it stated radar read error

dc4472af9d9d3fc5|2023-03-17--09-52-58--10
Radar read error at start before I engaged Open Pilot but seemed to have sorted itself out for the rest of the ride and at 10:03 I get a check blind spot monitoring error on car but not comma 3 unit.

sshane, 2023-03-22:
It looks like this might be a problem with the panda, I checked and the radar CAN messages drop out (cease to be received) in that segment you posted above, causing openpilot to display a radar error. @briskspirit `totalErrorCnt` quickly rises when we no longer receive the radar messages, could this be a panda issue? I don't see any ACC fault codes that likely would have been thrown if the radar stopped sending messages.

[image]

briskspirit, 2023-03-22:
I would check cable and connections if it happens occasionally and can't be reproduced easily. Something might go weird with one of the diff pair wires for CAN1. That's why totalErrorCnt also goes up and it should fall into passive or busoff state.

jmsether, 2023-03-24:
I have checked the harness on the connector. I have taken it apart twice and reseated it. The error is almost a daily occurrence now. It's random with no easy conditions observed. Could it be the relay itself?

sshane, 2023-08-07:
@jmsether does this still happen?

jmsether, 2023-08-07:
@sshane 

I found what was causing the error. The harness is the perfect length to be an antenna for wifi. I was getting interference on the CAN bus from the harness. I wrapped the harness in some EMF shielding and my problem seems to be fixed. I am still investigating the issue at hand but for now, it seems to be fixed or mitigated.

sshane, 2023-08-08:
closing then, thanks for the update. cc: @briskspirit if this is concerning at all

closed 2023-08-08 by sshane