commaai/openpilot #27563: 2017 Honda CR-V (Bosch) All warning/error lights go off on cold start

jesse-lapin, 2023-03-11:
### Describe the bug

When I start my car sometimes all of the warning/error lights in the dash go off and there is  an error on the C3. It happens about 10-20% of the time

### Which car does this affect?

2017 Honda CR-V EX-L

### Provide a route where the issue occurs

701ee2843363d318|2023-03-08--18-16-42

### openpilot version

master (0.9.12)

### Additional info

It also happened on 701ee2843363d318|2023-03-08--16-55-51. Here is a picture of the dash and the boot logs from that day. I believe it even wipes out the memory for the drivers seat position. 
[OP Bug.zip](https://github.com/commaai/openpilot/files/10948786/OP.Bug.zip)

paulvay, 2023-03-22:
I experience the same thing recently. I think this was after updating software.

On Honda CRV 2019 EX-L
Ver: 0.9.1-release3-cc18662.
Comma3
Honda Bosch harness purchased in 2021 for use with comma2.

recovery is achieved by turning car off and then back on.

sshane, 2023-03-28:
The panda is logging a lot of forwarding errors. Does this happen if you replace the OBD-C cable (short USB-C cable)?

jesse-lapin, 2023-03-28:
Any specifications on what type of usb-c cable is needed?

sshane, 2023-03-28:
Looks like some messages may be getting dropped sending from OP. Does this happen if you replace the OBD-C cable (short USB-C cable)? A high quality USB C 3.1 should be good to test.

jesse-lapin, 2023-04-01:
I bought 2 new USBC 3.2 cables from Amazon, both give this error every
time, so the cable might be the problem. Do you have a specific cable you
recommend? I kept the cable from my Comma 2 when I did the trade in and
never installed the new C3 cable, were there different cable specs?

jesse-lapin, 2023-07-29:
Hi, this issue has been occurring more and more frequently with the new
cable. It makes me think the USB plug on the C3 itself is faulty. I'm past
the 1 year for the warranty but I was hoping I could get a replacement
since I reported the issue while under warranty.

adeebshihadeh, 2023-07-31:
I'll reopen this. @sshane can you take a look?

sshane, 2023-07-31:
@jesse-lapin do you have a more recent route where this happens?

jesse-lapin, 2023-07-31:
It happened on this drive: 701ee2843363d318|2023-07-30--10-59-44--0

sshane, 2023-08-01:
Not too sure what's going on here, but the EPS enters a perm fault status on your first route (but not the latest route), so could be multiple things going on here. On the first route (701ee2843363d318|2023-03-08--16-55-51), it seems fine until we start sending the steering command and totalRxLostCnt rises on canState2.

[image]

jesse-lapin, 2023-08-01:
It only happens when I first start the car. Most of the time turning it off
and on again fixes it but not always. After buying the new cable it didn't
happen for maybe a month. It has been becoming increasingly frequent over
the last month or so.

-Jesse

jesse-lapin, 2023-08-13:
Hi, any update on this?

Thanks,
Jesse

jesse-lapin, 2023-08-18:
Hi, following up again, thank you.

jesse-lapin, 2023-08-21:
Could I send it in for a discount on a 3x? I've been a customer for years
and had the c2 before my C3 which I purchased before the price drop.
Thanks.

sshane, 2023-08-24:
@briskspirit can you check this out?

briskspirit, 2023-08-24:
can we get a route on latest master build? One I see here has older can_health packet.

jesse-lapin, 2023-08-24:
It's been happening on every 0.9x build I believe. I'll try to get the next
one.

jesse-lapin, 2023-09-01:
Here it is again this morning. 701ee2843363d318|2023-09-01--07-30-45--0

briskspirit, 2023-09-01:
You are using Sunny's fork. If you want us to look into your problem - switch to our master branch as I requested earlier.

jesse-lapin, 2023-09-01:
Ok, I'll switch, I've done that before to reproduce it so I don't expect it
to be different.

sshane, 2023-09-07:
@jesse-lapin did you get a route?

jesse-lapin, 2023-09-08:
Hi Shane, I was on a road trip for the holiday and didn't want to go back
to stock. Just flashed it tonight.

jesse-lapin, 2023-09-08:
It happened again this morning.

701ee2843363d318|2023-09-08--07-34-23--0

jesse-lapin, 2023-09-14:
Any update here? It keeps happening.

jesse-lapin, 2023-09-28:
Hi, it has been 2 weeks since I sent the latest drive with the official
build installed. Any update here?

Thanks,
Jesse

sshane, 2023-09-29:
@briskspirit Got a new route on master: `701ee2843363d318|2023-09-08--07-34-23`. See a few errors on startup (not too many), but certain messages drop off the bus around when it switches OBD multiplexing

briskspirit, 2023-09-29:
@sshane I don't see any weird behavior from the panda side. You multiplex and get can errors in process of doing it - that's normal. No high IRQ rates, no abnormal msg losses.

sshane, 2023-09-29:
You can see that at least all labeled as coming from the VSA module except one (`BRAKE_MODULE`) stopped sending not on multiplex change, but slightly after:

[image]

@jesse-lapin it might be worth seeing if your dealer can take a look at a faulty VSM module, since 1) Igor reports no abnormalities with the panda, 2) you see the dash faults while we're fingerprinting, so we're not sending any control messages yet, and 3) the EPS permanently faults a few seconds after which isn't normal.

jesse-lapin, 2023-09-30:
My car isn't under warranty anymore and it seems suspicious that changing
the cable made the error happen less frequently if the issue was in the car
itself.

Deltic9009, 2024-01-30:
I’ve been having the same problem. Every three or four weeks, on startup, every possible warning flashes up and cycles through. After around 20 minutes the system starts working normally.
I’ve read through the thread but don’t understand most of what I’ve read. Can someone explain in lay terms what the problem is and is there a fix.
I have a 2018 C-RV and been using open pilot for three or four months.
Thanks

sshane, 2024-03-20:
@jesse-lapin going to close since you haven't been using the device recently. If this still happens, let us know.

sshane, 2024-03-20:
@Deltic9009 does this still happen?

closed 2024-03-20 by sshane

jesse-lapin, 2024-03-20:
I traded in the CRV for a '24 Telluride.