commaai/openpilot #27575: C3 intermittently unable to connect to wifi networks

jacyhung, 2023-03-14:
### Describe the bug

From time to time, my C3 refuses to connect to any wifi networks. This is only the third time this has happened, otherwise, it connects just fine. C3 can see wifi networks just fine as well with full strength in UI. Attached are the dmesg logs I pulled off serial, but should also be visible as boot logs in my useradmin as I still have my SIM in (albeit super, super slow). I should also note that I have been unable to connect to the wifi hotspot created when tethering, which normally works as well.

[dmesg-wlan.txt](https://github.com/commaai/openpilot/files/10962833/dmesg-wlan.txt)
[dmesg.txt](https://github.com/commaai/openpilot/files/10962834/dmesg.txt)
[dmesg when attempting to join weedle (unsuccessful) ](https://gist.githubusercontent.com/jacyhung/c52fe52305b2947ec4a4d1495d3d974f/raw/f41a9a9d8935a96b62e04dcd4a04424996497ce7/gistfile1.txt)

### Provide a route where the issue occurs

d015f0d097b9e23a

### openpilot version

0.9.2

adeebshihadeh, 2023-09-02:
I see this device is decommissioned. Still having issues with the new one?

jacyhung, 2023-09-03:
> I see this device is decommissioned. Still having issues with the new one?

No issues after SOM was replaced.

closed 2023-09-03 by adeebshihadeh