commaai/openpilot #27628: Bolt EUV poor experimental ACC performance

ncoop23, 2023-03-20:
### Describe the bug

Have a log here of very poor ACC performance. 
0def4a390f6fe5c0|2023-03-19--19-01-38--0
Many times it stopped or went incredibly slow on straight roads outside a subdivision. Roads had poorly marked lines but asphalt in decent condition. 

### Provide a route where the issue occurs

0def4a390f6fe5c0|2023-03-19--19-01-38--0

### openpilot version

0.9.2 Mar 18

### Additional info

Can record more routes if needed. I don't think the car has radar.

morrislee, 2023-03-21:
there is a reason that it is experimental, we all experience slow acceleration at this time. this is like the first release of experimental long, and barely anyone has a solid model to tune the controls too. we will have to wait a bit unfortunately....

ncoop23, 2023-03-21:
No worries. I added this to possibly add in the images for training where the car struggled.

sshane, 2023-08-08:
controls looks fine, so going to close since it's likely more related to the model than the Bolt