commaai/openpilot #27661: Getting car unrecognized message on supported car

sdvolks12, 2023-03-24:
### Describe the bug

[image]

### Which car does this affect?

2019 Volkswagen Arteon SEL-P

### Provide a route where the issue occurs

https://connect.comma.ai/32011e33e8b1b837/1679613721742/1679616063256

### openpilot version

0.9.1

sshane, 2023-03-24:
Can you try this branch? https://github.com/commaai/openpilot/pull/27671

Installer URL would be: https://installer.comma.ai/commaai/vw-arteon-fp

sdvolks12, 2023-03-24:
Installing now

sdvolks12, 2023-03-24:
[image]

sdvolks12, 2023-03-24:
[image]

It stays on this screen

sshane, 2023-03-24:
Try to start the comma three while the car is off, and let it flash the panda. Then it should work.

sdvolks12, 2023-03-24:
Works now thank you!

sshane, 2023-03-25:
Merged! You're welcome to switch to any of these branches to stay up to date: `master`, `master-ci`, or `nightly`

closed 2023-03-25 by sshane