commaai/openpilot #27664: Nissan Leaf 2020 unrecognized

hammer983, 2023-03-24:
### Describe the bug

I've taken my Comma 3 out my Lexus and installed it in a Nissan Leaf (with the Leaf car harness) however I can't get it to work at all. I get the messages seat belt not fastened then car not supported and it switches to dash cam mode

### Which car does this affect?

Nissan Leaf 2020

### Provide a route where the issue occurs

5023d5087dde6adf|2023-03-23--20-44-08--0

### openpilot version

0.9.1

sshane, 2023-03-24:
Hi @hammer983. Can you try this branch? https://github.com/commaai/openpilot/pull/27670

For the installer URL you'll want to type in https://installer.comma.ai/commaai/nissan-fp

For the simplification of the Nissan port, can you get a route where you put your seatbelt on and off a few times? Make sure the car is on with the comma three on as well (showing camera).

sshane, 2023-03-25:
@hammer983 I saw you tried the branch but didn't engage. Were there any problems?

hammer983, 2023-03-26:
> @hammer983 I saw you tried the branch but didn't engage. Were there any problems?

Hi Shane, I did reply via email but not sure if you got the reply. I went for a short drive yesterday and it seemed all good. I put the seat belt on/off a few times and the warning came up on the screen.

Do I stick with this branch or reinstall the stock?

sshane, 2023-03-27:
Great, can you put your device on WiFi to upload some logs?

sshane, 2023-03-27:
Merged your fingerprint into master, you can now check out `master-ci` until the next release to get the latest updates.

closed 2023-03-27 by sshane

hammer983, 2023-03-28:
I've just connected to the Wifi so should be uploading now

Thanks for the support, appreciate it